balena-io/etcher #3418: [Question]Somehow the website redirected me to an older version download link,how?

Valleriite, 2021-01-22:
- **Etcher version:** 
- **expected version:**1.5.115
- **downloaded version:**1.5.45
- **Operating system and architecture:** 
- ArchLinux x86_64
- **Image flashed:** 
- Not related
- **Do you see any meaningful error information in the DevTools?** 
- Not related

### Warning:links shared may be wrong/contain malicious software since I am suspecting that my PC has been compromised.
this only happened one time and the website redirected me to this:
https://github-production-release-asset-2e65be.s3.amazonaws.com/45055693/6fdc5980-86c2-11e9-8154-82af6fd0f988?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210120%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210120T120107Z&X-Amz-Expires=300&X-Amz-Signature=817973cc9058c713c451c6c3eb3834dbe62ba0265046df5e7c68a677f94bccd8&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=45055693&response-content-disposition=attachment%3B%20filename%3Dbalena-etcher-electron-1.5.45-linux-x64.zip&response-content-type=application%2Foctet-stream

when I retried it redirected to this
https://github-production-release-asset-2e65be.s3.amazonaws.com/45055693/caecbe00-5984-11eb-9708-2c45e5f6419a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210120%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210120T141426Z&X-Amz-Expires=300&X-Amz-Signature=664322ed2daee2222a70a3d9cf096cd67d1c99c55510796b8ad4ff57de4c0c8b&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=45055693&response-content-disposition=attachment%3B%20filename%3Dbalena-etcher-electron-1.5.115-linux-x64.zip&response-content-type=application%2Foctet-stream

Is that some GitHub/Balena website bug?Or is my PC somehow compromised?

thundron, 2021-01-22:
version 1.5.45 *was* the version downloadable when scripts are disabled on balena.io (it's a user config we have no control on), but we recently bumped that version to a more recent one.
I think it's likely that you either downloaded between the transition from the old to the new or when the website wasn't done checking if the scripts were enabled.
since we can't be sure which case it is (if any), if you suspect that you have malicious software running I suggest a deep scan anyway, just to be safe

closed 2021-01-22 by thundron

lurch, 2021-01-22:
See also e.g. #3347

Valleriite, 2021-01-22:
> version 1.5.45 _was_ the version downloadable when scripts are disabled on balena.io (it's a user config we have no control on), but we recently bumped that version to a more recent one.
> I think [...]
I am not sure if this helps but the download was done on Wednesday January 20th,when was the switch done?

Valleriite, 2021-01-22:
also I am sure I don't have scripts disabled,unless my browser somehow blocked it on your website

Valleriite, 2021-01-22:
@thundron
I was able to reproduce the 1.5.45 download by clicking the download button on the download list on the bottom
before the website has fully loaded so your theory of the downloading before the checking has finished may be correct.
also that means you may have not bumped the version to a newer one since I was able to download 1.5.45

thundron, 2021-01-22:
> I was able to reproduce the 1.5.45 download by clicking the download button on the download list on the bottom

you're right - the change didn't make it to upstream. good catch!