balena-io/etcher #3419: "Flash" from  SMB Share

m4t7e0, 2021-01-22:
- **Etcher version: 1.5.109** 
- **Operating system and architecture: Windows 10 Pro x64** 
- **Image flashed: Anyone latest was FydeOS** 

Unable lo "Flash" image for  Mounted SMB Share . 
Same image copied on localdrive is working fine!

lurch, 2021-01-22:
Duplicate of #3122 and #2858 ?

m4t7e0, 2021-01-22:
[image]
[image]
[image]

Check it

julian-fairfax, 2021-04-12:
Hi,
I have this same issue on Linux. I've discovered a workaround, which is to use cifs to mount it to a "local" folder, but it would be much easier if this feature was added to Etcher, because it would then be possible to just mount in in the Files app (or another file manager) without having to run any commands or use sudo. Thanks for your work on this wonderful app!

julian-fairfax, 2021-04-12:
I've just put the files on an external disk which solves the problem, but it'd still be nice to have this as a feature!