balena-io/etcher #3421: Does not work without sudo in arch

JamesTheCode, 2021-01-25:
- **Etcher version:** 1.5.113
- **Operating system and architecture:** archlinux x64
- **Image flashed:**  N/A
- **Do you see any meaningful error information in the DevTools?**  
- error Error: User did not grant permission.
    at ChildProcess.exithandler (child_process.js:304)
    at ChildProcess.emit (events.js:223)
    at maybeClose (internal/child_process.js:1021)
    at Socket.<anonymous> (internal/child_process.js:430)
    at Socket.emit (events.js:223)
    at Pipe.<anonymous> (net.js:664)

The app wont say anything it just resets the target device - once I knew what was the problem, it was easily solved in console with sudo

[-1611593637967.log](https://github.com/balena-io/etcher/files/5868087/-1611593637967.log)

zvin, 2021-01-28:
@JamesTheCode does `pkexec echo "it works"` work for you ?

JamesTheCode, 2021-01-28:
Not without graphical sudo prompt.

zvin, 2021-01-28:
@JamesTheCode 
That's the purpose of `pkexec`.
When you run `pkexec echo "it works"` does it show you a graphical sudo prompt and if yes, does it print `it works` after you type your password ?

JamesTheCode, 2021-01-28:
yes

zvin, 2021-01-28:
Can you try the AppImage ?

zvin, 2021-01-28:
@JamesTheCode 
We don't provide arch packages. Where did you get etcher from ?

You can try adding some debug in `/usr/lib/balena-etcher/generated/gui.js` around line 66215 (depends on the version).

Search for `error.message = PERMISSION_DENIED;` and add something like `console.log(error, stdout, stderr)` on the line before it.
You should see this message in the developer console when you press flash (`CTRL` + `SHIFT` + `i` to open it).

zvin, 2021-01-28:
I guess it's from https://aur.archlinux.org/packages/balena-etcher/

JamesTheCode, 2021-02-04:
Yes, exactly arch user repository

JamesTheCode, 2021-02-04:
I cant repeat the error, i tried adding the console.log, but it worked just fine even though the version did not change. 
Instead of resetting the target device, it asked for elevated permission and then it worked. So sorry for bothering you and if it happens next time, i'll get the extra info from gui.js and paste it here.

closed 2021-02-04 by JamesTheCode