balena-io/etcher #3422: E: The repository 'https://deb.etcher.io stable Release' is not signed.

swipswaps, 2021-01-28:
- **Etcher version:** 
- **Operating system and architecture:** 
MX Linux
Linux mx 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+mx17+1 (2019-11-17) x86_64 GNU/Linux
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

Tim-Hobbes, 2021-02-15:
You probably forgot to trust Bintray.com's GPG key. This can be done by running this command: `sudo apt-key adv --keyserver hkps://keyserver.ubuntu.com:443 --recv-keys 379CE192D401AB61`

thundron, 2021-02-16:
correct, duplicate of https://github.com/balena-io/etcher/issues/2187, closing

closed 2021-02-16 by thundron

Bkroland19, 2021-08-09:
didnt get a solution

lurch, 2021-08-10:
@Bkroland19 AFAIK https://deb.etcher.io shouldn't be used any more. See #3501