balena-io/etcher #3425: Can't flash an iso file to my drive?

Derayyy, 2021-01-31:
For some reason i couldn't flash my windows 10 ISO file into my drive, every time i would select the file, there would be a popup that would say, "Missing partition table" i'm kinda new to ISO files and stuff so maybe i did something wrong? I use Linux btw.
[image]

pat106, 2021-02-07:
Just answered, same reason

https://github.com/balena-io/etcher/issues/3426

mabras, 2021-02-15:
Hi @Derayyy, Did you fixed it somehow?

Thanks,

Derayyy, 2021-02-25:
@mabras Uh, no, i didn't really understand what the person was talking about? I'm like, super new to this :/

lurch, 2021-02-25:
> flash my windows 10 ISO file

See https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

Derayyy, 2021-02-26:
@lurch So, its not possible to flash my windows 10 iso file with Etcher? I tried downloading Rufus but it didn't work too.

lurch, 2021-02-26:
As I understand it (I'm only a community-member, not an Etcher-developer), the problem is that Etcher can _burn_ the Windows ISO okay (if you ignore Etcher's warning), but your PC will be unable to _boot_ from the USB drive afterwards. I've never tried burning the Windows 10 ISO myself, so I've never tried using Rufus.

mabras, 2021-02-26:
> @lurch So, its not possible to flash my windows 10 iso file with Etcher? I tried downloading Rufus but it didn't work too.

No, I have used `WoeUSB` in the end.

thundron, 2021-03-22:
correct, it's not possible to flash windows isos using etcher, or at least it's very discouraged because they will likely not work as they require additional steps

closed 2021-03-22 by thundron