balena-io/etcher #3426: Etcher Win 10 stuck on "Starting..." with no error

nheminger, 2021-02-01:
- **Etcher version:** 1.5.115
- **Operating system and architecture:** Win 10 x86
- **Image flashed:** CentOS-Userland-7-aarch64-RaspberryPI-Minimal-4-2009-sda.raw
- **Do you see any meaningful error information in the DevTools?** Last messages are: Auto blockmapping: true and Decompress first: true. Then it sits at "Starting..." forever with nothing happening.

If I cancel then the SD card I am writing to becomes unusable and it can't be reformatted using regular Win 10 formt and needs to be erased and partitioned using Powershell.

pat106, 2021-02-07:
I think you are confusing the issue with a normal outcome. The fact that you can see the drive on windows means its assigned some kind of FS (Fat32, exfat, ntfs,...etc) either in MBR or GPT, usually the images we flash linux are not in a FS windows can read, write, or even see in any builtin tools as the will end up un some version of EXT. What youre doing with powershell is exactly what etcher does to flash starting with a clean slate, which you can expect to finish the procxess with a windows request to format the drive and no info about the main partition where the data was decompresssed. 

If you launch etcher as admin, that saves the hassle of this or that.
I'm no developper, or expert, just a guy who's been dealing with learning the technical stuff, using the tried and true method of -I want this, so I spend that, but likely I just end up doing a better job by making all the mistakes in the world and taking 25x times the effort, but one day the accumulation of knowledge will put me on the DOGECOIN path because it's at 0.06$ and Elon told me to do it. Never question Elon, we Musk follow the teachings of the master

thundron, 2021-03-22:
The second part (Windows not recognizing the target) is expected, though the first one isn't.
Does it make a difference if you run it with admin privileges?