balena-io/etcher #3431: Etcher hangs in 'Starting' state

vedyilmaz, 2021-02-10:
- **Etcher version:1.5.116
- **Operating system and architecture: Windows 10 x64 
- **Image flashed:balena-cloud-balena-homeassistant-raspberrypi4-64-2.67.3+rev4-dev-v12.3.0.img 
- **Do you see any meaningful error information in the DevTools? No errors

vedyilmaz, 2021-02-10:
It is hanging in 'Starting' state not proceeding.

geira, 2021-04-01:
I just had a similar experience after installing 1.5.116 on MacOS Catalina 10.15.7. Application just hangs on startup, and while for a brief time I could see the menus, all I could get was "Application not responding" in the process manager. Some time later I discovered my firewall had interrupted several connections to balena.io, when I blocked these (and ingest.sentry.io) the application seemed to start normally.

I'm not generally a fan of "phone home" software, that's why I use open source whenever possible. If this is for the collection of statistics then the GDPR states you must obtain voluntary consent before logging. However, since this is a bug report after all the very least one should do is to add a reasonable timeout of e.g. 10 seconds before failing the connection so that the application may continue starting up normally.

dylankershaw, 2022-04-06:
I resolved with `sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher`

jimmiedave, 2026-10-11:
Just got this:

```
sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
Password:
(electron) 'session.getPreloads' is deprecated and will be removed. Please use 'session.getPreloadScripts' instead.
(electron) 'session.setPreloads' is deprecated and will be removed. Please use 'session.registerPreloadScript' instead.
Language changed to: en
ready-to-show: 319.029ms
packageUpdatable false
[90783:1011/162626.701509:ERROR:net/cert/internal/trust_store_mac.cc:817] Error parsing certificate:
ERROR: Failed parsing extensions
****
```

jimmiedave, 2026-10-11:
Just got a resolution with balenaEtcher 2.1.6, no terminal necessary for the fix.

Formatted thumb drive as MS-DOS (FAT32). (it was apparently APFS beforehand). 

No idea why this would matter. It shouldn't matter if you're about to nuke the thumb drive a bit at a time anyway. But it worked, after maybe 45 minutes of dicking around just trying to get Etcher to do the one thing it exists for.