balena-io/etcher #3434: Validating Flash - Etcher 1.5.116 - Something went wrong

tigerg2002, 2021-02-18:
- 1.5.116
- Windows 10 Version 20H2 up to date as of 2/18/2021
- 2021-01-11-raspios-buster-armhf-lite
- I don't see DevTools!
 
I just love the information error message "Something went wrong"! I successfully flashed this same image on another SD card with a 1.4 version of Etcher a few minutes ago. Then saw the Update Available message so updated. Now validation fails. 

Is it really hard to put at least a hint of information in the error message? Seems to be common in this day and age. When I was writing software years ago this would have been unacceptable. Oh how we have advanced backward.

kempy007, 2021-02-19:
Bump: BSOD on validation stage. Writes seem to complete and card boots.

tigerg2002, 2021-02-19:
Yes, the card is usable. To my point: the message is useless.

eadmaster, 2022-05-23:
same here with an old Kingston-branded driverless sd card adapter, reportedly it is a Windows 10 exfat driver error.