balena-io/etcher #3441: macOS_BigSur

poohomar, 2021-02-23:
- **Etcher version:** Version 1.5.116 (1.5.116)
- **Operating system and architecture:** macOS BigSur
- **Image flashed:** TrueNAS-12.0-U2.iso
- **Do you see any meaningful error information in the DevTools?** No

[image]

Hello,
I can upload the .iso, I can select the target, but when I click on "Flash" then I'm prompted for a password (it doesn't give me the normal macOS elevation dialogue where I enter the elevated username). Once I enter a password, nothing happens, it just goes back to the "Select Target" screen (see attachment)?

Please advise

Obverser, 2021-02-23:
Same issue ^

Obverser, 2021-02-23:
Are you signed in with a normal account? If you have access to an admin account--you'll need to sign in there to flash.

poohomar, 2021-02-23:
Hello Gabriel,

Much appreciated for the feedback, I am signed in with my normal (non-admin) account. I have access to the elevated account, but it's cumbersome to switch accounts to run an action I should be able to use an elevated account for. For example, if I'm in some form of IT training class (which I am) and I'm trying to perform classroom instructions by following along. It's difficult to explain to the instructor I need to switch accounts for a few minutes to perform one action, a function other students are doing on the fly. 

Thanks,
Omar

lurch, 2021-02-25:
This has been a problem since MacOS Catalina - see e.g. https://github.com/balena-io/etcher/search?q=catalina&type=issues for other issues where this was discussed.

poohomar, 2021-02-25:
Sooo, I don't know why I didn't think of this earlier (probably because I haven't had to do this on macOS in like 4+ years), but here is a work around from non-admin accounts (only tested in for the Flash function and "not" URL or Clone).

You have to launch Terminal and then go root# for your privileged account. Once I ran the below command I was able to navigate to the location of the .iso of my case. Once I selected the file, then Target, I clicked on Flash waiting for it to fail, LOL. But, to my amazement it flashed and did a validation check and you what do you know, it was complete.

root# /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher

[image]

poohomar, 2021-02-25:
Something else I noticed, once you leave root# - then the application closes.
When you open it back up using root#, I get this string (don't know about anyone else).
Hope this helps the developers?

(node:54396) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app/generated/modules/xxhash/build/Release/hash.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
(node:54396) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app/generated/modules/@balena.io/usb/build/Release/usb_bindings.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
ready-to-show: 2134.415ms
Checking for update
Error: Error: net::ERR_CERT_INVALID
    at SimpleURLLoaderWrapper.<anonymous> (electron/js2c/browser_init.js:93:6395)
    at SimpleURLLoaderWrapper.emit (events.js:223:5)
{"stack":"Error: net::ERR_CERT_INVALID\n    at SimpleURLLoaderWrapper.<anonymous> (electron/js2c/browser_init.js:93:6395)\n    at SimpleURLLoaderWrapper.emit (events.js:223:5)","message":"net::ERR_CERT_INVALID"}
{"stack":"Error: net::ERR_CERT_INVALID\n    at SimpleURLLoaderWrapper.<anonymous> (electron/js2c/browser_init.js:93:6395)\n    at SimpleURLLoaderWrapper.emit (events.js:223:5)","message":"net::ERR_CERT_INVALID"}

Obverser, 2021-02-25:
Good find! This seems to solve the problem here too.

Just for reference for others:
1. Open a terminal window (or your terminal emulator of choice)
1. Sign into a sudoer/elevated account
1. Run `root# /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher` in the terminal

thundron, 2021-03-01:
We have a possible solution [here](https://github.com/balena-io/etcher/issues/3321) on how to implement the native password prompt, otherwise it's best to use an admin account on macOS Catalina and above

closed 2021-03-01 by thundron