balena-io/etcher #3442: build windows package on macOS catalina failed

fenglang0203, 2021-02-25:
- **Etcher version:** 
- 1.5.115
- **Operating system and architecture:** 
- macOS catalina
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 
I built a Windows package on MacOS, but it was successfully built, but it could not run. The following error is reported.
 Can I build a Windows package on MacOS?
```
esources\app\generated\modules\@ronomon\direct-io\binding.node is not a valid Win32 application.
```

lurch, 2021-02-26:
I don't understand how the Etcher build process works (I think it's all done behind the scenes on Balena's private CI server?), but I think https://github.com/balena-io/etcher/blob/master/beforeBuild.js might be relevant to your problem?

thundron, 2021-03-01:
You should build in the environment of the package, in this case Windows, for it to correctly build native modules such as the one linked there. It would be interesting to have cross-platform compilation but I don't think it's possible yet or, at least, not without workarounds (e.g. using `wine` for Windows packages, changing some risky env variables, etc)

closed 2021-03-01 by thundron