balena-io/etcher #3451: Windows Etcher Flash fail

PsychoData, 2021-03-08:
- **Etcher version:** 1.5.116
- **Operating system and architecture:** Windows 10 Pro - 20H2 - Build 19042.804
- **Image flashed:** [hassos_intel-nuc-5.12.img.xz](https://github.com/home-assistant/operating-system/releases#:~:text=hassos_intel-nuc-5.12.img.xz)
- **Do you see any meaningful error information in the DevTools?** No, I had rebooted after the failed flash as a troubleshooting step

I tried to flash this [hassos_intel-nuc-5.12.img.xz](https://github.com/home-assistant/operating-system/releases#:~:text=hassos_intel-nuc-5.12.img.xz) image onto a SATA HDD over a USB-SATA adapter. 

I accidentally bumped a cord loose around 50% into the image, and the drive is weird now. I checked into the drive in DiskPart, and it looks like it is showing as 8MB now - where the actual drive is around 40GB - so something seems broken? 

[image]

I tried Cleaning the disk, as [this suggests](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#windows) but it doesn't seem to change anything, and it looks like a perfectly fine ... 8 MB ... SSD 

[image]

Any suggestions?

lurch, 2021-03-08:
That looks... very weird!

> Any suggestions?

Do you have a different USB-SATA adapter, or a motherboard with built-in SATA ports, that you could try connecting your HDD to? I assume you've already tried the "turn it off and on again"? :laughing:

PsychoData, 2021-03-09:
Haha yeah - I tried a reboot and it didn't seem to be recognizing on my built in SATA ports - but they might haver been Off or in RAID mode - I haven't gotten back into my BIOS to check yet. fully yet. 

I tried a different USB-SATA and it does the same thing in Windows. 

I'll try getting it hooked up to internal SATA directly, and see if Windows can see the full size to format - and if not I'll try Linux and dd'ing just like a stupid amount of zeroes onto it

thundron, 2021-03-12:
I doubt this has anything to do with Etcher unfortunately :/ if even Windows doesn't see the drive correctly, there is probably something else going on, maybe hardware-related since you bumped a cord.
If @jvmenthon suggestion doesn't work, maybe you could try other recovery tools or check the Windows forums (first result I see when searching on Google https://www.easeus.com/resource/disk-unknown-not-initialized.html#:~:text=If%20%22disk%20unknown%20not%20initialized,an%20MBR%20or%20GPT%20disk.&text=Thus%2C%20when%20it%20is%20corrupted,showing%20as%20%22Not%20Initialized%22. )

closed 2021-03-12 by thundron

PsychoData, 2021-03-16:
yeah I'm thinking hardware / bump cord related. 
Windows still didn't like it even over SATA - same problem - but I took the drive to a linux Distro and just wrong a couple GB of 0's to it with `dd` and then Windows was able to pick it up again fine, and I used balena to flash it fine, even through the USB Sata adapter - I just made sure not to jostle my table or cords while it wrote

I don't have the exact command I used handy, but it was something like `dd if=/dev/zero of=/dev/sda bs=1 count=2000000`