balena-io/etcher #3454: Trying to Install Debian version on RaspberryPi 4 With 8GB

rok6969, 2021-03-11:
I am following the directions (cut / pasting) as directed but I am getting the following issues on the following steps:

Debian and Ubuntu based Package Repository (GNU/Linux x86/x64)

I am not sure if this is an actual issue, or just informational
command
sudo apt-get update
response
Hit:1 http://packages.microsoft.com/repos/code stable InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease                  
Hit:3 http://raspbian.raspberrypi.org/raspbian buster InRelease               
Ign:4 https://dl.bintray.com/etcher/debian stable InRelease                   
Get:5 https://dl.bintray.com/etcher/debian stable Release [3,674 B]
Get:6 https://dl.bintray.com/etcher/debian stable Release.gpg [821 B]
Fetched 4,495 B in 5s (881 B/s)    
Reading package lists... Done
N: Skipping acquire of configured file 'etcher/binary-armhf/Packages' as repository 'https://deb.etcher.io stable InRelease' doesn't support architecture 'armhf'

command 
sudo apt-get install balena-etcher-electron
response
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package balena-etcher-electron

lurch, 2021-03-11:
Probably a duplicate of #2870

rok6969, 2021-03-11:
Andrew,

Thanks for the reply….

Does this mean the issue has been resolved? I was not really able to follow the convo thread fully. It seems like  maybe there was a solution, but others seemed hesitant to implement it.

Bob

lurch, 2021-03-11:
I'm not an Etcher developer (merely an Etcher "friend"), and I've not tried running Etcher on a Raspberry Pi myself, but AFAICT #2870 is saying that a 3rd-party user has compiled an older Etcher version into a `.deb` that you could install on your Raspberry Pi.
I think the fact that #2870 is still open means that people are still waiting for an "official" solution.

thundron, 2021-03-12:
please refer to the other issue for tracking on this matter.

n.b. we still have no concrete plans to distribute packages for arm

closed 2021-03-12 by thundron