balena-io/etcher #3455: Different version specified in download compared to application

danPudwellUK, 2021-03-11:
Downloaded the latest version from the website, v1.5.116 for MacOS
However after installing it says I have 1.4.4
[image]
[image]

lurch, 2021-03-11:
You must have both the old version and the new version installed, and somehow you're still running the old version. Note that the old version was "resin.io Etcher" but the new version is "balenaEtcher".

thundron, 2021-03-12:
as @lurch said, please check for old versions on your system and remove them if necessary

closed 2021-03-12 by thundron