balena-io/etcher #3459: Mac Big Sur - not starting write process

chbaines, 2021-03-18:
Version 1.5.116 (1.5.116)
Big Sur 11.2.3 macbook pro
Raspbian-Homebridge-v1.0.20.zip 
 
The logs after the pop up dialog asking permission to flash the SD drive is below. The process stops like it was cancelled even though the proper password was entered. Tried every trick, sudo, giving disk access, etc. nothing works, it just cancels out after the permissions dialog.

Wed Mar 17 2021 20:01:23 GMT-0500 (Central Daylight Time) SafeWebview loaded ({"event":{"id":61,"url":"https://assets.balena.io/etcher-featured/index.html?borderRight=false&darkBackground=true&etcher-version=1.5.116&api-version=2&optOutAnalytics=false","method":"GET","timestamp":1616029281925.0889,"resourceType":"mainFrame","ip":"104.21.29.149","fromCache":false,"statusLine":"HTTP/1.1 200","statusCode":200,"responseHeaders":{"access-control-allow-origin":["*"],"age":["420"],"alt-svc":["h3-27=\":443\"; ma=86400, h3-28=\":443\"; ma=86400, h3-29=\":443\"; ma=86400"],"cache-control":["max-age=600"],"cf-cache-status":["DYNAMIC"],"cf-ray":["631a8784cfd217f7-ATL"],"cf-request-id":["08e47306fd000017f72780b000000001"],"content-encoding":["br"],"content-type":["text/html; charset=utf-8"],"date":["Thu, 18 Mar 2021 01:01:22 GMT"],"expect-ct":["max-age=604800, report-uri=\"https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct\""],"expires":["Thu, 11 Mar 2021 10:16:02 GMT"],"last-modified":["Mon, 08 Feb 2021 16:56:46 GMT"],"nel":["{\"max_age\":604800,\"report_to\":\"cf-nel\"}"],"report-to":["{\"group\":\"cf-nel\",\"endpoints\":[{\"url\":\"https:\\/\\/a.nel.cloudflare.com\\/report?s=QVnYAeyZ%2BwwLVgL73tp%2F2KiNES8Tn1KTXqqZMW8JD4nQq%2FpxLvLq%2B8rM%2FXiMUTxHY0id96UwBQRMVb9Vbhv9lepyly1897sdtGPXqaI1ahDL\"}],\"max_age\":604800}"],"server":["cloudflare"],"status":["200"],"vary":["Accept-Encoding"],"via":["1.1 varnish"],"x-cache":["HIT"],"x-cache-hits":["2"],"x-fastly-request-id":["953d7cb59dc75cf66f6644683f898901b5891970"],"x-github-request-id":["9C04:221A:E7180:839ACE:6049EB8A"],"x-origin-cache":["HIT"],"x-proxy-cache":["MISS"],"x-served-by":["cache-pdk17880-PDK"],"x-timer":["S1616029282.056671,VS0,VE0"]},"referrer":"","error":"net::OK"},"sample":0.1,"applicationSessionUuid":"2ff6e624-d042-489f-8d71-7d8498f22519","flashingWorkflowUuid":"fca9245e-5654-4c38-8173-c4484d7600b8"})
gui.js:31760  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
gui.js:42734 Terminating IPC server
gui.js:42734 Flash results {cancelled: true, skip: false}
gui.js:42734 Wed Mar 17 2021 20:01:36 GMT-0500 (Central Daylight Time) Elevation cancelled ({"image":{"size":2492097024,"name":"Raspbian-Homebridge-2.img","hasMBR":true,"partitions":[{"offset":4194304,"size":262144000,"type":12,"index":1},{"offset":266338304,"size":2225758720,"type":131,"index":2}],"extension":"img","path":"/Users/chbaines/Downloads/Raspbian-Homebridge-2.img"},"drives":[{"size":7948206080,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/PCI0@0/XHC1@14","mountpoints":[{"path":"/Volumes/UNTITILED","label":"UNTITILED"}],"description":"Mass Storage Device Media","isUAS":null,"partitionTableType":"gpt"}],"driveCount":1,"uuid":"eb222d33-3aab-4c03-a843-c9a8f094a98b","status":"cancel","flashInstanceUuid":"eb222d33-3aab-4c03-a843-c9a8f094a98b","sample":0.1,"applicationSessionUuid":"2ff6e624-d042-489f-8d71-7d8498f22519","flashingWorkflowUuid":"fca9245e-5654-4c38-8173-c4484d7600b8"})
gui.js:42734 Wed Mar 17 2021 20:02:27 GMT-0500 (Central Daylight Time) Open external link ({"url":"https://github.com/balena-io/etcher/blob/master/SUPPORT.md","sample":0.1,"applicationSessionUuid":"2ff6e624-d042-489f-8d71-7d8498f22519","flashingWorkflowUuid":"fca9245e-5654-4c38-8173-c4484d7600b8"})

Hades32, 2021-04-22:
This is a duplicate of https://github.com/balena-io/etcher/issues/3441 . The current workaround is to run `sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher` in the terminal

closed 2021-04-22 by Hades32

llamafilm, 2021-12-15:
Almost the same symptom, but for me in #3651 it never prompts for a password, it just fails.  The `sudo` command above works for me.