balena-io/etcher #3460: Missing password prompt in v1.5.116 on macOS 10.15 (EACCES error)

mcous, 2021-03-18:
Hello! I use Etcher really frequently for work, and I noticed a strange change in behavior after upgrading from v1.5.115 to v1.5.116.

I have tried reinstalling the application directly from GitHub releases, but I have _not_ yet tried clearing out `~/Library/Application\ Support/balena-etcher/`. Downgrading back to `115` works around the issue described below.

## previous / expected behavior

1. Select file
2. Select target
3. Click "Flash"
4. Etcher pops up password prompt, enter password
5. Etcher flashes target

## new behavior

1. Select file
2. Select target
3. Click "Flash"
4. No password prompt, instead Etcher fails with an `EACCES` error

```json
{
  "stack": "Error: EACCES: permission denied, open '/var/folders/j5/_yv5y3hj3q39d98d0hzgn58r0000gn/T/etcher/balena-etcher-electron-438f2f51af7b.cmd'",
  "message": "EACCES: permission denied, open '/var/folders/j5/_yv5y3hj3q39d98d0hzgn58r0000gn/T/etcher/balena-etcher-electron-438f2f51af7b.cmd'",
  "errno": -13,
  "code": "EACCES",
  "syscall": "open",
  "path": "balena-etcher-electron-438f2f51af7b.cmd",
  "image": "ot2-fullimage(9).zip"
}
```

## application / system details

- v1.5.116
- macOS 10.15.7
- [Opentrons OT-2 full OS image](https://opentrons-buildroot-ci.s3.amazonaws.com/8cb98573-1d48-46e5-8566-3427986b6e2e/opentrons-buildroot/ot2-fullimage.zip)
- Don't think so, error message posted above

johnweber, 2021-05-13:
Confirmed that this issue seems to still exist on 1.5.120. A downgrade to 1.5.115 resolves the issue, but I didn't try any versions between 1.5.116 and 1.5.120.

    Etcher version: v1.5.120
    Operating system and architecture: macOS 10.15.7 x64
    Image flashed: [https://developer.nvidia.com/jetson-nano-sd-card-image](https://developer.nvidia.com/jetson-nano-sd-card-image)

mcous, 2023-07-21:
Closed due to lack of attention / staleness

closed 2023-07-21 by mcous