balena-io/etcher #3462: etcher not installing on Arch Linux

Enjoy-Maity, 2021-03-19:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

It takes forever and then the package build fails. It shows

[0:03:00] Still working on:
[0:03:00]   src/third_party/electron_node
[0:03:00]   src/chrome/test/data/xr/webvr_info
[0:03:00]   src/third_party/angle
[0:03:00]   src/third_party/boringssl/src
[0:03:00]   src/third_party/breakpad/breakpad
[0:03:00]   src/third_party/catapult
[0:03:00]   src/third_party/chromite
[0:03:00]   src/third_party/cld_3/src
 and keeps on going

lurch, 2021-03-19:
Sounds like a problem with the ArchLinux etcher-build-script, which IIRC is maintained by a 3rd party?

Enjoy-Maity, 2021-03-19:
> Sounds like a problem with the ArchLinux etcher-build-script, which IIRC is maintained by a 3rd party?

i actually don't know, I have tried installing balena etcher using yay , git clone 
and even tried installing different dependencies one by one before making the build, but every time it returns something like this.

lurch, 2021-03-19:
Have you tried using the AppImage instead?

Enjoy-Maity, 2021-03-19:
appimage is working fine

thundron, 2021-03-22:
If using the AppImage isn't feasible for any reason, I would suggest to refer to the owner of the distribution channel that you installed Etcher from as Arch is supported by the community

closed 2021-03-22 by thundron