balena-io/etcher #3467: Password not working

TheToastJester, 2021-03-27:
- **1.5.116** 
- **Mac OS:** 
- **retropie-buster-4.7.1-rpi4_400.img** 
- **Do you see any meaningful error information in the DevTools?** 

Asking for password and then not taking it I assume, it goes back to the original screen.

lurch, 2021-03-27:
You might need to try logging in with an admin account instead? I don't have a Mac, and I'm not an Etcher developer, but I've seen a number of people reporting [issues with Etcher on Catalina](https://github.com/balena-io/etcher/issues?q=is%3Aissue+catalina) and above.

TheToastJester, 2021-03-27:
Tried that. Catalina sucks

TheToastJester, 2021-03-30:
I manually transported the file from another user, couldn't get it flashed. Catalina's security measure caused an issue.

closed 2021-03-30 by TheToastJester