balena-io/etcher #3469: Flashing fails on drives with 300MB/s+ write speeds

guyish1, 2021-03-27:
- **Etcher version: 1.5.116** 
- **Operating system and architecture: windows 10 x86** 
- **Image flashed: ubuntu 20.04 an kali2021.1 live** 
- **Do you see any meaningful error information in the DevTools? no** 

I'm using a USB to data adapter and it seems that any drive faster than 300MB/s write speeds causes the flashing to fail, I tried flashing to a Samsung SM841 SATA SSD with 300MB/s writes and the flash files but then I use a SanDisk z400s with 180MB/s writes and it works.

[image]
[image]

KubanLink, 2021-03-28:
How much space do you have allocated to swap? and How much is allocated to Linux?

manimaddictedtolinux, 2021-04-05:
Think they should limit the speed or just fix it

manimaddictedtolinux, 2021-04-05:
it says the bus type is USB, but isUSB is false.

guyish1, 2021-04-05:
> 
> 
> it says the bus type is USB, but isUSB is false.

its a sata drive using a USB to sata adapter

manimaddictedtolinux, 2021-04-05:
> > it says the bus type is USB, but isUSB is false.
> 
> its a sata drive using a USB to sata adapter

Oh, well try limit the speed then.

lurch, 2021-04-06:
I guess using a USB2 port (or a USB2 hub plugged into a USB3 port) might be one way of artificially "limiting the speed"? :shrug: