balena-io/etcher #3474: Possible performance issue with flashing with bmap

jcormier, 2021-03-29:
- **Etcher version:** 1.5.80
- **Operating system and architecture:** Windows 10 x86_64
- **Image flashed:** Custom image created by Yocto
- **Do you see any meaningful error information in the DevTools?** No

With the bmap enabled zip, etcher reports it wrote ~6.1G instead of ~10.3GB which is good.  But it seems at least on my machine,  the original zip averaged 31MB/s write speed, but the bmap enabled zip only averaged 11MB/s.

And so the bmap enabled zip took longer...

Bmap zip trials: 606s, 659s, 661s
Old zip trials: 356s, 350s

Also got a consistent checksum error for the last block of data:
Checksum does not match for range [11048124416, 11048136703]: "84ff92691f909a05b224e1c56abb4864f01b4f8e3c854e4bb4c7baf1d3f6d652" != "0f6a85c0f9e90c4bfbb623a9489953d0b19841ca35807d23250324f57d0a2cea"

But the image itself passed a md5sum check.

lurch, 2021-03-29:
> Etcher version: 1.5.80

That's a fairly old version?

jcormier, 2021-03-29:
> Also got a consistent checksum error for the last block of data:
> Checksum does not match for range [11048124416, 11048136703]: "84ff92691f909a05b224e1c56abb4864f01b4f8e3c854e4bb4c7baf1d3f6d652" != "0f6a85c0f9e90c4bfbb623a9489953d0b19841ca35807d23250324f57d0a2cea"

Source image file matches bmap file:

```
<Range chksum="84ff92691f909a05b224e1c56abb4864f01b4f8e3c854e4bb4c7baf1d3f6d652"> 2697296-2697298 </Range>
```

```
jcormier@LAPTOP-JCORMIER2012:/mnt/c/Users/jcormier/Downloads$ dd if=tisdk-rootfs-image-mitysom-am57x-20210324193841.rootfs.img bs=4096 skip=2697296 count=3 | sha256sum
2+1 records in
2+1 records out
84ff92691f909a05b224e1c56abb4864f01b4f8e3c854e4bb4c7baf1d3f6d652  -
10240 bytes (10 kB, 10 KiB) copied, 0.0006457 s, 15.9 MB/s
```

Flashed sd card matches etchers calculation:
```
root@mitysom-am57x:~# dd if=/dev/sdb bs=4096 skip=2697296 count=3 | sha256sum
3+0 records in
3+0 records out
0f6a85c0f9e90c4bfbb623a9489953d0b19841ca35807d23250324f57d0a2cea  -
```

Looks like the image file doesn't end on a block boundary, looks like this is an issue with bmaptool.  It can't address image files that aren't a multiple of its chosen block size.
```
root@mitysom-am57x:~# dd if=/dev/sdb bs=4096 skip=2697296 count=3 | hexdump -C
3+0 records in
3+0 records out
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00002800  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00003000
root@mitysom-am57x:~# logout
Connection to 192.168.0.29 closed.
jcormier@LAPTOP-JCORMIER2012:/mnt/c/Users/jcormier/Downloads$ dd if=tisdk-rootfs-image-mitysom-am57x-20210324193841.rootfs.img bs=4096 skip=2697296 count=3 | hexdump -C
2+1 records in
2+1 records out
10240 bytes (10 kB, 10 KiB) copied, 0.0005108 s, 20.0 MB/s
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00002800
```

jcormier, 2021-03-29:
> That's a fairly old version?

Fair, I've had auto-updates enabled so I just assumed it was up to date...  I can retest on latest

closed 2021-03-29 by jcormier