balena-io/etcher #3477: Feature request: BeagleBone flashing

jadonk, 2021-04-01:
I see that etcher-sdk 6.2.1 now includes BeagleBone boot support (https://github.com/balena-io-modules/etcher-sdk/commit/2a1d745bf59ca93739f489d7ae85ba19bc2697da). It seems the current release of Etcher is still using etcher-sdk 5.2.2 (https://github.com/balena-io/etcher/commit/37769efbeda0abe7993d95e2b2aea2f461edd307).

Is there a plan to update to the latest etcher-sdk and to add the GUI components to perform flashing of BeagleBones?

zvin, 2021-04-02:
@jadonk This is not in Etcher yet because we need to get it working on Windows.

jadonk, 2021-04-02:
Do you have any way to reproduce errors? Know what needs to be done?

zwhitchcox, 2021-04-02:
@jadonk basically no one has ever flashed a Beagle Bone from Windows, so we do not have a POC. Once we have that, we will be able to integrate it into Etcher.

The code we used to as a POC for Etcher and windows is [here](https://github.com/ravikp7/node-beagle-boot) and [here]https://github.com/parthi-g/node-beaglebone-usbboot#readme).

We can get Beaglebone bootloader to load U-Boot, and but Windows does not recognize it as a UMS.

We think the problem might be that the version of U-Boot is 7 years old, which presents a few problems. Specifically, the version of GadgetFS is fairly old, and the new U-Boot version includes a new configuration for Beagle Bone Black, which is not in the old one.

I was working on this, but I got pulled into another project. I think we are going to give this investigation info to the BB foundation and see if they can produce a working version that can be flashed from Windows.

jadonk, 2021-04-08:
Using u-boot as the UMS host is really a hack to try to simplify the work of reproducing the image a tad easier. What might be better is to bootload a small BalenaOS image and run UMS off that. UMS in Linux is much faster than UMS in u-boot.

I'm with BeagleBoard.org Foundation. I could probably provide you with a Buildroot image to boot, but I suspect BalenaOS images might be easier for you to maintain. You'd simply want to boot one that used a RAM-based rootfs and enabled the appropriate Linux kernel gadget drivers mounted to the right volume locations for flashing.

lurch, 2021-04-09:
@jadonk IIRC that's exactly the approach used by https://github.com/balena-io-modules/node-raspberrypi-usbboot/tree/master/blobs/raspberrypi

zwhitchcox, 2021-04-09:
@zvin Do you have a recommended approach for this?

jadonk, 2021-04-14:
@zvin Are you able to build a BalenaOS image that boots with a ramdisk, the right gadget driver and exposes the /dev/mmcblkX devices? I've done something like this before, but I do not personally have time for this unless you'd want to pay me. @parthi-g might be able to do this.

lurch, 2021-04-14:
@jadonk That was originally being done (for the Raspberry Pi) in https://github.com/balena-os/amber-etcher-kernel but I've got no idea where that code might have moved to now.

zwhitchcox, 2021-04-14:
@lurch We are still using that same kernel...it hasn't changed

lurch, 2021-04-14:
@zwhitchcox Ahh cool. I only assumed that the code now "lived" elsewhere because the repo I linked to is marked as read-only.

shawaj, 2021-05-29:
@zwhitchcox has there been any progress on this?

And is it possible to build a version for Linux and macOS that can program BBB?

Do you have any build instructions by any chance? Or even better, a packaged app?

shawaj, 2021-07-20:
ping @zwhitchcox @jadonk @zvin ?

jadonk, 2021-07-20:
@parthi put together a build at one point. I will update a ping to some Balena folks to provide a Linux ramdisk image suitable for Windows UMS support.

zwhitchcox, 2021-07-21:
Hey, @jadonk @parthi-g actually never got it to work on Windows, nor did @ravikp7

I still think the easiest solution is going to be just updating U-boot.

zwhitchcox, 2021-07-21:
hey, @jadonk

>I could probably provide you with a Buildroot image to boot, 

If you could provide a repo with instructions/a script to build the image, I think we could pretty much maintain/modify it. I think this would be the absolute fastest way to get BB support into Etcher, and the users are clamoring for it.

I think we will later on down the road try to create a BalenaOS image, but we are in the middle of a major change to Etcher right now that I think we will want to get done first.

Is this suitable for you?

jadonk, 2021-07-22:
@parthi--g and @ravikp7 both used U-Boot, rather than Linux, for UMS. I'm struggling to get on IRC to check with any U-Boot developers if the UMS in U-Boot has gotten any faster or support for Windows has improved.

Asking me to do the Buildroot image is definitely taking the slow route as I don't have much free time. I'll look into it.

zwhitchcox, 2021-08-23:
Able to boot beagle bone now, but unable to get windows to automatically select the correct usb interface...will update later.