balena-io/etcher #3479: Unable to install on debian bullseye with README instructions

danrneal, 2021-04-03:
- **Operating system and architecture:** 
Debian testing (Bullseye)

```
echo "deb https://deb.etcher.io stable etcher" | sudo tee /etc/apt/sources.list.d/balena-etcher.list
sudo apt-key adv --keyserver hkps://keyserver.ubuntu.com:443 --recv-keys 379CE192D401AB61
sudo apt-get update
sudo apt-get install balena-etcher-electron

Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 balena-etcher-electron : Depends: libappindicator1 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

lurch, 2021-04-03:
Duplicate of #3406 which was apparently fixed in #3394 ?
It might be that the version in the apt repo still predates that, so you could try downloading the latest deb from https://github.com/balena-io/etcher/releases

thundron, 2021-04-30:
please see https://github.com/balena-io/etcher/issues/3501