balena-io/etcher #3481: ethcer has broken disk I flashed to

godalming123, 2021-04-12:
- **1.5.116** 
- **windows 64bit** 
- **https://download.fedoraproject.org/pub/fedora/linux/releases/test/34_Beta/Workstation/x86_64/iso/Fedora-Workstation-Live-x86_64-34_Beta-1.3.iso** 
- ** _____ _       _
    |    ___|   |     | |
    |   |__   | |_ ___| |__   ___ _ __
    |    __|  | __/ __| '_ \ / _ \ '__|
    |   |___  | || (__| | | |  __/ |
    \____/ \__\___|_| |_|\___|_|
    
    Interested in joining the Etcher team?
    Drop us a line at join+etcher@balena.io
    
    Version = 1.5.116, Type = nsis
    gui.js:52086 Mon Apr 12 2021 14:21:43 GMT+0100 (British Summer Time) Application start ({"packageType":"nsis","version":"1.5.116","sample":0.1,"applicationSessionUuid":"d47d1765-26a4-44b4-8465-181e9279da13","flashingWorkflowUuid":"1c72e84c-bf84-4dc4-8df0-682f5e819c49"})
    gui.js:42737 Mon Apr 12 2021 14:21:46 GMT+0100 (British Summer Time) Open image URL selector ({"sample":0.1,"applicationSessionUuid":"d47d1765-26a4-44b4-8465-181e9279da13","flashingWorkflowUuid":"1c72e84c-bf84-4dc4-8df0-682f5e819c49"})
    gui.js:42737 Mon Apr 12 2021 14:21:53 GMT+0100 (British Summer Time) Select image ({"image":{"size":2001469440,"name":"Fedora-Workstation-Live-x86_64-34_Beta-1.3.iso","hasMBR":true,"partitions":[{"offset":88064,"size":11509760,"type":239,"index":1}],"extension":"iso","path":"https://download.fedoraproject.org/pub/fedora/linux/releases/test/34_Beta/Workstation/x86_64/iso/Fedora-Workstation-Live-x86_64-34_Beta-1.3.iso","logo":false,"blockMap":false},"sample":0.1,"applicationSessionUuid":"d47d1765-26a4-44b4-8465-181e9279da13","flashingWorkflowUuid":"1c72e84c-bf84-4dc4-8df0-682f5e819c49"})
    gui.js:42737 Mon Apr 12 2021 14:21:56 GMT+0100 (British Summer Time) Flash ({"image":{"size":2001469440,"name":"Fedora-Workstation-Live-x86_64-34_Beta-1.3.iso","hasMBR":true,"partitions":[{"offset":88064,"size":11509760,"type":239,"index":1}],"extension":"iso","path":"https://download.fedoraproject.org/pub/fedora/linux/releases/test/34_Beta/Workstation/x86_64/iso/Fedora-Workstation-Live-x86_64-34_Beta-1.3.iso"},"drives":[{"size":30943995904,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive2","error":null,"isReadOnly":false,"displayName":"E:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive2","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"E:\\"}],"description":"Kingston DataTraveler 3.0 USB Device","isUAS":false,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"0629623e-2685-47fd-8354-ca28362292ab","status":"started","flashInstanceUuid":"0629623e-2685-47fd-8354-ca28362292ab","sample":0.1,"applicationSessionUuid":"d47d1765-26a4-44b4-8465-181e9279da13","flashingWorkflowUuid":"1c72e84c-bf84-4dc4-8df0-682f5e819c49"})
    gui.js:42737 Elevating command: C:\Users\jamie\AppData\Local\Programs\balena-etcher\balenaEtcher.exe C:\Users\jamie\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js
    gui.js:31760  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
    gui.js:42737 Mon Apr 12 2021 14:21:57 GMT+0100 (British Summer Time) SafeWebview loaded ({"event":{"id":1,"url":"https://assets.balena.io/etcher-featured/index.html?borderRight=false&darkBackground=true&etcher-version=1.5.116&api-version=2&optOutAnalytics=false","method":"GET","timestamp":1618233717115.001,"resourceType":"mainFrame","ip":"2606:4700:3034::6815:1d95","fromCache":false,"statusLine":"HTTP/1.1 200","statusCode":200,"responseHeaders":{"access-control-allow-origin":["*"],"age":["402"],"alt-svc":["h3-27=\":443\"; ma=86400, h3-28=\":443\"; ma=86400, h3-29=\":443\"; ma=86400"],"cache-control":["max-age=600"],"cf-cache-status":["DYNAMIC"],"cf-ray":["63ed1aa8dcbe2c8a-LHR"],"cf-request-id":["09680efd8700002c8ad32e4000000001"],"content-encoding":["br"],"content-type":["text/html; charset=utf-8"],"date":["Mon, 12 Apr 2021 14:21:58 GMT"],"expect-ct":["max-age=604800, report-uri=\"https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct\""],"expires":["Mon, 12 Apr 2021 05:44:36 GMT"],"last-modified":["Fri, 02 Apr 2021 21:01:06 GMT"],"nel":["{\"max_age\":604800,\"report_to\":\"cf-nel\"}"],"report-to":["{\"endpoints\":[{\"url\":\"https:\\/\\/a.nel.cloudflare.com\\/report?s=g3eJjOzhnLC3lbob4Y3n0AA0N3URUV%2F8SZtiCY9bCA7KYZiHCUc9Nv4z6yYu%2BOhMqB%2F1ZFRF85lBe5tA%2BhZGUHKKnsTZ5pxyFY4om3fidZpC2pajs6Pk0mpVvVax\"}],\"group\":\"cf-nel\",\"max_age\":604800}"],"server":["cloudflare"],"status":["200"],"vary":["Accept-Encoding"],"via":["1.1 varnish"],"x-cache":["HIT"],"x-cache-hits":["2"],"x-fastly-request-id":["5795e7d3e9d93959268f83ba3c00990d76058d0c"],"x-github-request-id":["692A:6732:29EB8F:2EF4BA:6073DBEC"],"x-origin-cache":["HIT"],"x-proxy-cache":["MISS"],"x-served-by":["cache-lon4238-LON"],"x-timer":["S1618237319.555868,VS0,VE0"]},"referrer":"","error":"net::OK"},"sample":0.1,"applicationSessionUuid":"d47d1765-26a4-44b4-8465-181e9279da13","flashingWorkflowUuid":"1c72e84c-bf84-4dc4-8df0-682f5e819c49"})
    gui.js:42737 Successfully connected to IPC server: etcher-server-3648, socket root C:\Users\jamie\AppData\Local\Temp\
    gui.js:42737 Image: https://download.fedoraproject.org/pub/fedora/linux/releases/test/34_Beta/Workstation/x86_64/iso/Fedora-Workstation-Live-x86_64-34_Beta-1.3.iso
    gui.js:42737 Devices: \\.\PhysicalDrive2
    gui.js:42737 Auto blockmapping: true
    gui.js:42737 Decompress first: true
    gui.js:42737 Mon Apr 12 2021 14:22:33 GMT+0100 (British Summer Time) Flash error ({"image":{"size":2001469440,"name":"Fedora-Workstation-Live-x86_64-34_Beta-1.3.iso","hasMBR":true,"partitions":[{"offset":88064,"size":11509760,"type":239,"index":1}],"extension":"iso","path":"https://download.fedoraproject.org/pub/fedora/linux/releases/test/34_Beta/Workstation/x86_64/iso/Fedora-Workstation-Live-x86_64-34_Beta-1.3.iso"},"drives":[{"size":30943995904,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive2","error":null,"isReadOnly":false,"displayName":"E:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive2","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"E:\\"}],"description":"Kingston DataTraveler 3.0 USB Device","isUAS":false,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"0629623e-2685-47fd-8354-ca28362292ab","flashInstanceUuid":"0629623e-2685-47fd-8354-ca28362292ab","error":{"name":"Error","message":"Couldn't clean the drive, Command failed: diskpart s n (code 2147942405)","stack":"Error: Couldn't clean the drive, Command failed: diskpart s n (code 2147942405)\n    at Object.clean (C:\\Users\\jamie\\AppData\\Local\\Programs\\balena-etcher\\resources\\app\\generated\\child-writer.js:8253:35)\n    at async BlockDevice._open (C:\\Users\\jamie\\AppData\\Local\\Programs\\balena-etcher\\resources\\app\\generated\\child-writer.js:9071:25)\n    at async BlockDevice.open (C:\\Users\\jamie\\AppData\\Local\\Programs\\balena-etcher\\resources\\app\\generated\\child-writer.js:10665:25)\n    at async n    at async Promise.all (index 0)\n    at async MultiDestination._open (C:\\Users\\jamie\\AppData\\Local\\Programs\\balena-etcher\\resources\\app\\generated\\child-writer.js:10300:21)\n    at async MultiDestination.open (C:\\Users\\jamie\\AppData\\Local\\Programs\\balena-etcher\\resources\\app\\generated\\child-writer.js:10665:25)\n    at async Promise.all (index 1)\n    at async pipeSourceToDestinations (C:\\Users\\jamie\\AppData\\Local\\Programs\\balena-etcher\\resources\\app\\generated\\child-writer.js:8490:17)\n    at async decompressThenFlash (C:\\Users\\jamie\\AppData\\Local\\Programs\\balena-etcher\\resources\\app\\generated\\child-writer.js:8426:28)"},"sample":0.1,"applicationSessionUuid":"d47d1765-26a4-44b4-8465-181e9279da13","flashingWorkflowUuid":"1c72e84c-bf84-4dc4-8df0-682f5e819c49"})
    gui.js:42737 Finish: 0
    gui.js:42737 Terminating IPC server
    gui.js:42737 Flash results Object
    gui.js:42737 Mon Apr 12 2021 14:22:34 GMT+0100 (British Summer Time) Done ({"image":{"size":2001469440,"name":"Fedora-Workstation-Live-x86_64-34_Beta-1.3.iso","hasMBR":true,"partitions":[{"offset":88064,"size":11509760,"type":239,"index":1}],"extension":"iso","path":"https://download.fedoraproject.org/pub/fedora/linux/releases/test/34_Beta/Workstation/x86_64/iso/Fedora-Workstation-Live-x86_64-34_Beta-1.3.iso"},"drives":[{"size":30943995904,"isVirtual":false,"enumerato [... cut ...] ","flashingWorkflowUuid":"1c72e84c-bf84-4dc4-8df0-682f5e819c49"})** 

Flashing with etcher has worked before the only deference this time is I'm flashing onto a usb that had a working copy of zorin os flashed on it. I have tried using diskpart as instructed by https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#windows but this has resulted in error:
```DISKPART> clean
DiskPart has encountered an error: Access is denied.
See the System Event Log for more information.
```
.

godalming123, 2021-04-26:
flashing works after deleteing the zorin system partition and replacing the unused space with a new volume.

thundron, 2021-04-30:
seems like an issue with the OS rather than etcher

closed 2021-04-30 by thundron