balena-io/etcher #3482: Debian/Ubuntu repository respond with error : invalid key

dominix, 2021-04-12:
- **Etcher version:** (last)
- **Operating system and architecture:**  Ubuntu LTS 20.04 focal
- **Image flashed:** not related
- **Do you see any meaningful error information in the DevTools?**  not related

sudo apt update 
...
Err :1 https://dl.bintray.com/etcher/debian stable InRelease                                
  403  Forbidden [IP : 52.42.217.145 443]
... 

Reading package lists... Done                      
E: Failed to fetch https://deb.etcher.io/dists/stable/InRelease  403  Forbidden [IP: 35.161.90.47 443]
E: The repository 'https://deb.etcher.io stable InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.cloud.google.com/apt cloud-sdk InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY FEEA9169307EA071 NO_PUBKEY 8B57C5C2836F4BEB

odinb, 2021-04-12:
Well, I am only getting:
Err:5 https://dl.bintray.com/etcher/debian stable InRelease
  403  Forbidden [IP: 52.42.240.241 443]

Seems they probably have issues with their hoster again! This seems to be a recurring issue every few months! Time to get a new hosting location?

pvanhemmen, 2021-04-26:
I can confirm this issue. Receving update information through the apt package manager is no longer possible due to failing signature verification.

thundron, 2021-04-30:
please see https://github.com/balena-io/etcher/issues/3501

closed 2021-04-30 by thundron