balena-io/etcher #3484: Blue screen after image validation

Climberirw, 2021-04-15:
- **Etcher version: 1.5.116** 
- **Operating system and architecture: Windows 10 x64** 
- **Image flashed: openwrt-19.07.7-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz (This also happens with any other image I flash)** 
- **Do you see any meaningful error information in the DevTools?**  - No

Whenever I try to flash an image (any image), everything seems normal until the image validation is finished, then my PC crashes and gives me a blue screen of death (the error code is different each time, but the most recent one was KMODE_EXCEPTION_NOT_HANDLED). Whenever I test out the MicroSD card, it's fine and the flash happened correctly.

lurch, 2021-04-15:
A BSoD often means a driver error. Do you have a different SD card-reader you could try using?

Connor-Irwin, 2021-04-15:
> A BSoD often means a driver error. Do you have a different SD card-reader you could try using?

I do, but the one I am using is built into my laptop so there shouldn't be any driver issues. I also didn't get a BSoD until I updated etcher to 1.5.116

lurch, 2021-04-15:
> the one I am using is built into my laptop so there shouldn't be any driver issues.

Not necessarily, see https://github.com/raspberrypi/rpi-imager/issues/94

thundron, 2021-04-16:
as @lurch mentioned a BSoD is often due to driver issues. the fact it happens regardless of the validation is likely to be attributed to releasing/closing the resources after the flash that, for some reason, doesn't go well with your current drivers. I suggest looking up for the BSoD error on Window's support as there's not much we can do in this case, unfortunately

closed 2021-04-16 by thundron

AubsUK, 2021-06-26:
FYI... I've just added a solution (at least for me) to the thread I started at https://github.com/raspberrypi/rpi-imager/issues/94 

This is a built-in card reader in a Dell laptop.