balena-io/etcher #3485: Success banner keeps reloading infinitely

gediz, 2021-04-16:
- **Etcher version:** Both v1.5.116 and v1.5.113
- **Operating system and architecture:** Ubuntu 20.04.2 LTS / x86_64
- **Image flashed:** Any valid image
- **Do you see any meaningful error information in the DevTools?**  It just loops over the following message:
`gui.js:56875 Fri Apr 16 2021 08:35:12 GMT+0300 (GMT+03:00) SafeWebview loaded ({"event":{"id":39,"url":"https://www.balena.io/etcher/success-banner/?borderTop=false&darkBackground=true&etcher-version=1.5.113&api-version=2&optOutAnalytics=false","method":"GET","timestamp":1618551312518.617,"resourceType":"mainFrame","ip":"18.159.128.50","fromCache":false,"statusLine":"HTTP/1.1 200","statusCode":200,"responseHeaders":{"age":["41894"],"cache-control":["public, max-age=0, must-revalidate"],"content-encoding":["br"],"content-length":["6047"],"content-type":["text/html; charset=UTF-8"],"date":["Thu, 15 Apr 2021 17:56:58 GMT"],"etag":["\"307fc85c003b6ae4fcb05c667c9a6511-ssl-df\""],"link":["</webpack-runtime-f4d4a0bcb0a64af36de6.js>; rel=preload; as=script, </styles-c4fa86dfd4ad05097eee.js>; rel=preload; as=script, </framework-42f2600cafd3c0e08aac.js>; rel=preload; as=script, </50c25a04-efe577ee14c9c53bed03.js>; rel=preload; as=script, </73ad3401-e112b47ad7a992b91ac8.js>; rel=preload; as=script, </532a2f07-16f3498dd087ee7288d4.js>; rel=preload; as=script, </app-c4121afa58c543c5ae66.js>; rel=preload; as=script, </c943faba-f859f80ceb7f4f7eab31.js>; rel=preload; as=script, </af13d906-36fdbc23fc7dd2f2b2e7.js>; rel=preload; as=script, </ec1189df-f1a321eba12a2bb0c7ab.js>; rel=preload; as=script, </a9a7754c-29508ff4045f1f93a5f6.js>; rel=preload; as=script, </commons-f107fce4e0358d11a866.js>; rel=preload; as=script, </component---src-pages-etcher-success-banner-js-b3df158ccec49c4b44f3.js>; rel=preload; as=script, </page-data/app-data.json>; rel=preload; as=fetch; crossorigin, </page-data/etcher/success-banner/page-data.json>; rel=preload; as=fetch; crossorigin"],"referrer-policy":["strict-origin-when-cross-origin"],"server":["Netlify"],"status":["200"],"strict-transport-security":["max-age=31536000"],"vary":["Accept-Encoding"],"x-content-type-options":["nosniff"],"x-frame-options":["DENY"],"x-nf-request-id":["c0d29256-f384-4318-a889-2bdfe52041bf-83291971"],"x-xss-protection":["1; mode=block"]},"referrer":"https://www.balena.io/etcher/success-banner/?borderTop=false&darkBackground=true&etcher-version=1.5.113&api-version=2&optOutAnalytics=false","error":"net::OK"},"sample":0.1,"applicationSessionUuid":"3f74c5a5-27ab-4966-b277-3d9ba079097b","flashingWorkflowUuid":"c8cb991d-7129-4107-9d4c-1fc89fc83b3a"})`

It's just a cosmetic error.

[image]

I've clicked the etcher logo on the page and it redirected me to https://www.balena.io/etcher/?ref=etcher_footer and this page keeps looping, too.

[image]

lurch, 2021-04-16:
>  https://www.balena.io/etcher/?ref=etcher_footer and this page keeps looping, too.

Hmmm, that does the same for me too, this is what I see in the Developer console (after ticking "preserve log")
```
Navigated to https://www.balena.io/etcher/?ref=etcher_footer
index.js:94 Could not fetch latest release, falling back to the preloaded one TypeError: Failed to fetch
2Navigated to https://www.balena.io/etcher/?ref=etcher_footer
?ref=etcher_footer:1 Access to XMLHttpRequest at 'https://data.balena-cloud.com/amplitude' from origin 'https://www.balena.io' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.
50c25a04-efe577ee14c9c53bed03.js:sourcemap:2 POST https://data.balena-cloud.com/amplitude net::ERR_FAILED
Ne.send @ 50c25a04-efe577ee14c9c53bed03.js:2
Pe.sendEvents @ 50c25a04-efe577ee14c9c53bed03.js:2
Pe._sendEventsIfReady @ 50c25a04-efe577ee14c9c53bed03.js:2
u @ 50c25a04-efe577ee14c9c53bed03.js:2
Pe.init @ 50c25a04-efe577ee14c9c53bed03.js:2
e @ app-c4121afa58c543c5ae66.js:2
t.createClient @ app-c4121afa58c543c5ae66.js:2
s @ app-c4121afa58c543c5ae66.js:2
GddB.t.onClientEntry @ app-c4121afa58c543c5ae66.js:2
(anonymous) @ app-c4121afa58c543c5ae66.js:2
Promise.then (async)
(anonymous) @ app-c4121afa58c543c5ae66.js:2
t.apiRunnerAsync @ app-c4121afa58c543c5ae66.js:2
UxWs @ app-c4121afa58c543c5ae66.js:2
d @ webpack-runtime-f4d4a0bcb0a64af36de6.js:1
a @ webpack-runtime-f4d4a0bcb0a64af36de6.js:1
c @ webpack-runtime-f4d4a0bcb0a64af36de6.js:1
(anonymous) @ framework-42f2600cafd3c0e08aac.js:2
Navigated to https://www.balena.io/etcher/?ref=etcher_footer
component---src-pages-etcher-index-js-ddddcdc31f3a94f03959.js:sourcemap:1 Could not fetch latest release, falling back to the preloaded one TypeError: Failed to fetch
Navigated to https://www.balena.io/etcher/?ref=etcher_footer
component---src-pages-etcher-index-js-ddddcdc31f3a94f03959.js:sourcemap:1 Could not fetch latest release, falling back to the preloaded one TypeError: Failed to fetch
2Navigated to https://www.balena.io/etcher/?ref=etcher_footer
?ref=etcher_footer:1 Access to XMLHttpRequest at 'https://data.balena-cloud.com/amplitude' from origin 'https://www.balena.io' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.
amplitude.js:1896 POST https://data.balena-cloud.com/amplitude net::ERR_FAILED
Ne.send @ 50c25a04-efe577ee14c9c53bed03.js:2
Pe.sendEvents @ 50c25a04-efe577ee14c9c53bed03.js:2
Pe._sendEventsIfReady @ 50c25a04-efe577ee14c9c53bed03.js:2
u @ 50c25a04-efe577ee14c9c53bed03.js:2
Pe.init @ 50c25a04-efe577ee14c9c53bed03.js:2
e @ app-c4121afa58c543c5ae66.js:2
t.createClient @ app-c4121afa58c543c5ae66.js:2
s @ app-c4121afa58c543c5ae66.js:2
GddB.t.onClientEntry @ app-c4121afa58c543c5ae66.js:2
(anonymous) @ app-c4121afa58c543c5ae66.js:2
Promise.then (async)
(anonymous) @ app-c4121afa58c543c5ae66.js:2
t.apiRunnerAsync @ app-c4121afa58c543c5ae66.js:2
UxWs @ app-c4121afa58c543c5ae66.js:2
d @ webpack-runtime-f4d4a0bcb0a64af36de6.js:1
a @ webpack-runtime-f4d4a0bcb0a64af36de6.js:1
c @ webpack-runtime-f4d4a0bcb0a64af36de6.js:1
(anonymous) @ framework-42f2600cafd3c0e08aac.js:2
Navigated to https://www.balena.io/etcher/?ref=etcher_footer
```

thundron, 2021-04-16:
It seems that is has to do with additional parameters as balena.io/etcher doesn't reload instead, we're investigating

gediz, 2021-04-19:
Looks like it's resolved.

closed 2021-04-19 by gediz