balena-io/etcher #3486: Ubuntu install orphaned path

emanuel4you, 2021-04-17:
...
Setting up balena-etcher-electron (1.5.117) ...
chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox': No such file or directory

kb2ma, 2021-05-03:
Yes, the `postinst` file is out of sync with the filesystem.

```
kbee@system76:~/tmp/etcher/control$ cat postinst
#!/bin/bash

# Link to the binary
ln -sf '/opt/balena-etcher-electron/balena-etcher-electron' '/usr/bin/balena-etcher-electron'

# SUID chrome-sandbox for Electron 5+
chmod 4755 '/opt/balena-etcher-electron/chrome-sandbox' || true

update-mime-database /usr/share/mime || true
update-desktop-database /usr/share/applications || true
```

lurch, 2021-05-03:
@thundron I'm also seeing this same error after following the instructions in #3051 (I'm running Ubuntu 18.04 LTS)
```
Need to get 86.8 MB of archives.
After this operation, 4,291 kB disk space will be freed.
Get:1 https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu bionic/main amd64 balena-etcher-electron amd64 1.5.117 [86.8 MB]
Fetched 86.8 MB in 7s (12.3 MB/s)                                              
(Reading database ... 589767 files and directories currently installed.)
Preparing to unpack .../balena-etcher-electron_1.5.117_amd64.deb ...
Unpacking balena-etcher-electron (1.5.117) over (1.5.100) ...
Setting up balena-etcher-electron (1.5.117) ...
chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox': No such file or directory
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for mime-support (3.60ubuntu1) ...
Processing triggers for desktop-file-utils (0.23-1ubuntu3.18.04.2) ...
Processing triggers for gnome-menus (3.13.3-11ubuntu1.1) ...
```
However if I `ls /opt` I see that there **is** a `/opt/balenaEtcher` directory? And 
```
$ ls -l /usr/bin/balena-etcher-electron 
lrwxrwxrwx 1 root root 50 May  3 08:10 /usr/bin/balena-etcher-electron -> /opt/balena-etcher-electron/balena-etcher-electron
```
is a broken symlink :confused:

M3LiNdRu, 2021-05-04:
It looks like there is something wrong with the path. Following commands provided by @kb2ma above using ` /opt/balenaEtcher `as directory works properly:

```
# Link to the binary
ln -sf '/opt/balenaEtcher/balena-etcher-electron' '/usr/bin/balena-etcher-electron'

# SUID chrome-sandbox for Electron 5+
chmod 4755 '/opt/balenaEtcher/chrome-sandbox' || true

update-mime-database /usr/share/mime || true
update-desktop-database /usr/share/applications || true
```

I guess this path is build using some sort of convention and the convention is broken.

lurch, 2021-05-04:
No need to ask my permission! I don't work on this project, I'm just an "interested 3rd party" :wink:

M3LiNdRu, 2021-05-05:
duplicated issue: https://github.com/balena-io/etcher/issues/3074

fix: https://github.com/balena-io/etcher/commit/d8cb8f78154910f46b70f4b2537d57169b1a0b60

lurch, 2021-05-05:
Seems strange that a bug apparently fixed in Feb 2020 would still be affecting Etcher in May 2021? :confused: :man_shrugging:

valdominic, 2021-05-24:
Is this issue fixed?

M3LiNdRu, 2021-05-25:
> 
> 
> Is this issue fixed?

It looks like the fix was released. However, .deb package still have the issue, so not sure what release is packaged inside the current .deb package.

@valdominic following the above steps provided by @kb2ma should fix the issue in my Debian

lurch, 2021-05-25:
@M3LiNdRu I've just taken a look at the issue you linked to, and I think that's a similar-but-different error. #3074 seems to be about a bug in the script used to _start_ Etcher, whereas this issue is about a bug in the package used to _install_ Etcher?

valdominic, 2021-05-28:
> > Is this issue fixed?
> 
> It looks like the fix was released. However, .deb package still have the issue, so not sure what release is packaged inside the current .deb package.
> 
> @valdominic following the above steps provided by @kb2ma should fix the issue in my Debian

@M3LiNdRu , when i try to run the fix that @kb2ma  has provided above, it gives me an error

Here is the error

bash: /home/debianbox/tmp/etcher/control$: No such file or directory

Eratosthenes, 2021-07-06:
I'm having this issue as well FYI.

KernelDeimos, 2021-12-01:
How is this _still_ broken?

cerosilva, 2021-12-06:
Not sure, but i just tried uninstalling and reinstalling it again and still have getting the same 'chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox': No such file or directory' error. 

Could this be the reason why my bootable USB is not working at boot time? It's showing up in explorer and i can work with the files. but it's not booting up my system with the loaded Distro.

theofficialgman, 2022-03-23:
bumping this because the control file is STILL broken

do-nat, 2022-12-11:
I just tried to install the apt package and this issue is still occuring. Is nobody working on this?

M3LiNdRu, 2022-12-12:
> I just tried to install the apt package and this issue is still occuring. Is nobody working on this?

@do-nat What linux distribution are you using?

FYI, not reproduced on Debian 11

do-nat, 2022-12-14:
I'm using PopOS, which is an Ubuntu derivate

suparious, 2023-03-05:
I just got this issue on a default install of Ubuntu 22.10

pjgoodall, 2025-05-27:
here we are in 2025 - and still getting this error!

lurch, 2025-09-18:
> Any other alternatives people would recommend?

You could try [Raspberry Pi Imager](https://www.raspberrypi.com/software/) ? It has a "select custom image" option.

seb-1204, 2026-10-25:
Hi, 
I’m seeing the same `chrome-sandbox` path mismatch with balenaEtcher 2.1.9 on Ubuntu 26.04.

During installation, the package reports:

```bash
chmod: cannot access '/opt/balenaEtcher/chrome-sandbox': No such file or directory
```

But the helper is installed here:

```bash
/usr/lib/balena-etcher/chrome-sandbox
```

Could the package’s setup script be updated to use the installed path, or otherwise locate the helper correctly? It seems odd to me that this issue persist over years.

suparious, 2026-10-31:
This is from 2021. The fact that you're posting here means that you just don't understand how things work.
Sandboxes are stupid. Using Google Chrome is stupid.

This issue persists because there are some people that don't know how to configure their operating system environments, and expect software package maintainers to build workarounds for their ignorance.

PR #3735 (2022-03-31): Fix application directory for Debian postinst script
Add an after-install template file to the configuration to fix the `postinst` script for the Debian package. See electron-builder [LinuxSpecificTargetOptions](https://www.electron.build/generated/linuxtargetspecificoptions). Must hardcode the application directory name in the script because there is not a variable available with the content of 'balenaEtcher', as you can see in the [app-builder-lib source](https://github.com/electron-userland/electron-builder/blob/24e986e664f89b8a41f4e5032709a73d30667f9d/packages/app-builder-lib/src/targets/f [... cut]