balena-io/etcher #3497: Trying to install etcher on my pi4 with raspbian/buster OS, error

Schvin7, 2021-04-25:
pi:~ $ sudo apt-get install balena-etcher-electron
Reading package lists... Done
Building dependency tree       
Reading state information... Done
**E: Unable to locate package balena-etcher-electron**

That's the error I keep getting. Started with a fresh install of Raspbian. I did sudo apt-get update and upgrade. Followed the steps on the github page, and that's the error I keep getting.    (Downloaded the "balena-etcher-electron-1.5.116-linux-x64.zip" version for my raspberry pi 4b 8gb)

thanks

lurch, 2021-04-26:
Duplicate of #2870 ?

Schvin7, 2021-04-26:
I apologize, I'm very new to this. I tried searching before I posted this thread. Everything I've found, I have tried, and failed.  I installed ruby gems last night, but fpm wouldnt install properly.  I'm kind of stuck. But i was able to download/install the Raspberry Pi Imager, so I can at least use that.

closed 2023-07-12 by dfunckt