balena-io/etcher #3498: BSOD when flashing hiveos-0.6-203-stable@210403.img.xz

matcarfer, 2021-04-26:
- **Etcher version:** balenaEtcher-Portable-1.5.117.exe
- **Operating system and architecture:** Win 10 Pro x64 20H2
- **Image flashed:** hiveos-0.6-203-stable@210403.img.xz
- **Do you see any meaningful error information in the DevTools?** Nope, as I had a BSOD

Nirsoft BlueScreenView log:

p	26/4/2021 19:09:26	KMODE_EXCEPTION_NOT_HANDLED	0x0000001e	ffffffff`c0000005	fffff801`4c06f403	00000000`00000000	00000000`00000040	volmgr.sys	volmgr.sys+f403	Controlador del administrador de volumen	Sistema operativo Microsoft® Windows®	Microsoft Corporation	10.0.19041.1 (WinBuild.160101.0800)	x64	ntoskrnl.exe+3f5e40					C:\Windows\Minidump\042621-10703-01.dmp	6	15	19041	1.725.044	26/4/2021 19:10:13	

ntoskrnl.exe	ntoskrnl.exe+49177f	fffff801`47200000	fffff801`48246000	0x01046000	0x0ea0877a		Microsoft® Windows® Operating System	NT Kernel & System	10.0.19041.928 (WinBuild.160101.0800)	Microsoft Corporation	C:\Windows\system32\ntoskrnl.exe	

volmgr.sys	volmgr.sys+f403	fffff801`4c060000	fffff801`4c079000	0x00019000	0x51e4251d	15/7/2013 13:36:45	Sistema operativo Microsoft® Windows®	Controlador del administrador de volumen	10.0.19041.1 (WinBuild.160101.0800)	Microsoft Corporation	C:\Windows\system32\drivers\volmgr.sys	

Attached MiniDump
[042621-10703-01.zip](https://github.com/balena-io/etcher/files/6380676/042621-10703-01.zip)

lurch, 2021-04-27:
Possibly a Windows driver error, do you have a different card-reader you can use?

matcarfer, 2021-04-27:
Not a card reader, it was a pendrive usb 3.0.

thundron, 2021-05-01:
I would suggest to look for the errors on windows forums, unfortunately there's not much we can do about BSODs as they mean your system is failing, we just happen to use the resources that are faulty already

closed 2021-05-01 by thundron

lurch, 2021-05-01:
@matcarfer It might be worth seeing if e.g. Win32DiskImager gives you the same error? (although you'll obviously need to decompress the image first as I don't think Win32DiskImager supports compressed files)

matcarfer, 2021-07-26:
Ok, just got another BSOD after (many) uses. 

Etcher version: balenaEtcher-Portable-1.5.121.exe
Operating system and architecture: Win 10 Pro x64 21H1
Image flashed: recalbox-x86_64.img.xz
Do you see any meaningful error information in the DevTools? Nope, as I had a BSOD.

Nirsoft Bluescreen Log:
072621-9312-01.dmp	26/7/2021 00:33:18	KMODE_EXCEPTION_NOT_HANDLED	0x0000001e	ffffffff`c0000005	fffff802`27abf403	00000000`00000000	00000000`00000040	volmgr.sys	volmgr.sys+f403	Controlador del administrador de volumen	Sistema operativo Microsoft® Windows®	Microsoft Corporation	10.0.19041.1 (WinBuild.160101.0800)	x64	ntoskrnl.exe+3f6f20					C:\Windows\Minidump\072621-9312-01.dmp	6	15	19041	2.481.892	26/7/2021 00:34:08	
[072621-9312-01.zip](https://github.com/balena-io/etcher/files/6875390/072621-9312-01.zip)

ntoskrnl.exe	ntoskrnl.exe+459c05	fffff802`22c00000	fffff802`23c46000	0x01046000	0x553df94c	27/4/2015 05:54:36	Microsoft® Windows® Operating System	NT Kernel & System	10.0.19041.1110 (WinBuild.160101.0800)	Microsoft Corporation	C:\Windows\system32\ntoskrnl.exe	

volmgr.sys	volmgr.sys+f403	fffff802`27ab0000	fffff802`27ac9000	0x00019000	0x51e4251d	15/7/2013 13:36:45	Sistema operativo Microsoft® Windows®	Controlador del administrador de volumen	10.0.19041.1 (WinBuild.160101.0800)	Microsoft Corporation	C:\Windows\system32\drivers\volmgr.sys	

I never EVER got any BSOD with Premiere, Games, etc. Just Balena Etcher. All I can say the process already started and it was flashing.