balena-io/etcher #3504: Failed to fetch https://deb.etcher.io/dists/stable/InRelease  403  Forbidden

BertLindeman, 2021-05-03:
- **Etcher version:**
not relevant 
- **Operating system and architecture:** 
Linux Mint 20.1
- **Image flashed:**
not relevant 
- **Do you see any meaningful error information in the DevTools?** 
Do not get that far as install fails

Did the advised action on #3501 and the README:
```
add new PPA:

curl -1sLf \
      'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
      | sudo -E bash
```

I assume there is still some redirection on the website, as I see:
```
Err:6 https://dl.bintray.com/etcher/debian stable InRelease
  403  Forbidden [IP: 3.124.41.190 443]
```
In a browser going to ```https://deb.etcher.io``` still redirects to:
```
https://dl.bintray.com/etcher/debian/
```
with the page showing ```forbidden```

lurch, 2021-05-03:
I guess you need to check your various `/etc/apt/sources.list` and `/etc/apt/sources.list.d/*.list` files to remove any stale references to `https://deb.etcher.io`, and then run `sudo apt update` ?  (for me that was done automatically by the `setup.deb.sh`, but maybe it doesn't always work in all cases?)

BertLindeman, 2021-05-03:
Thanks, Andrew for the fast response.

**conclusion**:
remove any sources.list entry for balena-etcher **before** running `setup.deb.sh`

**TL;DR**

Retry this time:
deleted `/sources.list.d/balena-etcher.list` 
Ran `apt update`  and redo:
```
curl -1sLf \
      'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
      | sudo -E bash
```

with this result:
```
Executing the setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=linuxmint  version=20.1  codename=ulyssa  arch=x86_64
 >>>>:
 NOPE: Checking for apt dependency 'apt-transport-https' ...
   OK: Updating apt repository metadata cache ...
   OK: Attempting to install 'apt-transport-https' ...
   OK: Checking for apt dependency 'ca-certificates' ...
   OK: Checking for apt dependency 'gnupg' ...
   OK: Importing 'balena/etcher' repository GPG key into apt ...
   OK: Checking if upstream install config is OK ...
   OK: Installing 'balena/etcher' repository via apt ...
   OK: Updating apt repository metadata cache ...
   OK: The repository has been installed successfully - You're ready to rock!
```
`apt update` runs fine now.

So conclusion:
remove any sources.list entry for balena-etcher **before** running `setup.deb.sh`

Thanks again, @lurch 

Closing.

closed 2021-05-03 by BertLindeman