balena-io/etcher #3505: only v1.5.116 (not latest 1.5.119) is available for download

ksdomino, 2021-05-03:
How can I get the latest version, or is it not available for mac?

lurch, 2021-05-03:
Looks like v1.5.117 is the latest version for which binaries are available: https://github.com/balena-io/etcher/releases

luckman212, 2021-05-03:
Came to ask the same, seems something is wrong with the build pipeline since 117...

lurch, 2021-05-03:
Perhaps none of the changes in v1.5.118 and v1.5.119 would actually affect the binaries that people download? :man_shrugging: 
https://github.com/balena-io/etcher/blob/master/CHANGELOG.md

ksdomino, 2021-05-03:
Old version on website (116 vs 119) so that's quite a bit of disparity: As of today, the version available for download is 3 months old and it looks like there have been quite a lot of changes since then so it really needs fixing but can't make demands, just pointing out the disparity.

thundron, 2021-05-03:
thanks for reporting, indeed there seems to be an issue on our CI when publishing the file that we then use to check the version on the website, we're investigating

thundron, 2021-05-17:
@lurch correct, closing!

closed 2021-05-17 by thundron