balena-io/etcher #3506: Broken link-shortcuts on website

lurch, 2021-05-03:
Near the middle of https://www.balena.io/etcher/ there's a bit that says "Looking for Debian (.deb) packages or Red Hat (.rpm) packages?" which has a link pointing to `https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64?d_id=ad0052e3-4bf5-4907-a61c-a175746f08f6R` (with similar for RPM).

However the query-part `?` of the URL stops the shortcut-part '#' of the URL working, and so the browser doesn't auto-scroll to the right part of the README. If I remove the query-part to get `https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64` then my browser _does_ auto-scroll to the right part of the page.

jelabarre59, 2021-09-10:
If we are bringing up broken links, the instructions for adding the repository for DNF (Fedora 34 in my case) set up the repo file to point to https://dl.cloudsmith.io/public/balena/etcher/rpm/fedora/34/$basearch.  No updated version of Etcher was found, so I checked the site.   That URL doesn't work (excluding the $basearch part, obviously), and navigating all the way back to the server root (https://dl.cloudsmith.io/) returns "Not Found     Could not find the requested resource" for every subdirectory I tried.

lurch, 2021-09-10:
@jelabarre59 https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/bionic/InRelease "works", but any "parent part" of that URL e.g. https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/bionic/ reports "Not Found", so I guess that's just the way the CloudSmith server is set up?

thundron, 2021-09-13:
thanks @lurch we'll look into solving it!

lurch, 2021-09-13:
The link is now `https://github.com/balena-io/etcher?d_id=ad0052e3-4bf5-4907-a61c-a175746f08f6R#debian-and-ubuntu-based-package-repository-gnulinux-x86x64` which means that the link-shortcuts are working now :tada:

closed 2021-09-13 by lurch

Zorono, 2022-03-09:
The same problem is occurring with Debian bullseye... (https://dl.cloudsmith.io/public/balena/etcher/deb/debian)

lurch, 2022-03-09:
@Zorono See #3716 ?