balena-io/etcher #3507: New repo doesn't work on Kubuntu 20.04

gmipf, 2021-05-03:
Here is the output:

```
:~$ curl -1sLf    'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh'    | sudo -E bash
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=ubuntu  version=20.04  codename=focal  arch=x86_64  
 >>>>:
   OK: Checking for apt dependency 'apt-transport-https' ...
   OK: Checking for apt dependency 'ca-certificates' ...
   OK: Checking for apt dependency 'gnupg' ...
 NOPE: Importing 'balena/etcher' repository GPG key into apt ...
 ^^^^: ... Could not import the GPG key for this repository
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact Cloudsmith (support@cloudsmith.io) for further assistance.
 !!!!: 
 !!!!: Last 20 log lines from .csm_setup_9Dx2UN8nq (might not be errors, nor even relevant):
 !!!!:
 !!!!: > Warning: apt-key output should not be parsed (stdout is not a terminal)
 !!!!: > gpg: invalid key resource URL /etc/apt/trusted.gpg.d/home:Provessor.gpg
 !!!!: > gpg: keyblock resource (null): General error
 !!!!: > gpg: key 76F1A20FF987672F: 1 signature not checked due to a missing key
 !!!!: > gpg: key 876292BF7F9C32BF: 1 signature not checked due to a missing key
 !!!!: > gpg: key 3B4FE6ACC0B21F32: 3 signatures not checked due to missing keys
 !!!!: > gpg: key D94AA3F0EFE21092: 3 signatures not checked due to missing keys
 !!!!: > gpg: key 871920D1991BC93C: 1 signature not checked due to a missing key
 !!!!: > gpg: Total number processed: 17
 !!!!: > gpg:       skipped new keys: 17
```

fcastilloec, 2021-05-03:
This error might have something to do with what we're seeing when trying to update using the official Debian repository.

```
E: Failed to fetch https://deb.etcher.io/dists/stable/InRelease  403  Forbidden [IP: 52.42.38.157 443]
E: The repository 'https://deb.etcher.io stable InRelease' is no longer signed.
```

Something is going on with their GPG key.

BertLindeman, 2021-05-03:
Maybe related to #3504 ?

I needed to clear up the previous ```sources.list``` entry for balena to get it right.

gmipf, 2021-05-03:
@BertLindeman I have already done that before adding the new repo. 

@fcastilloec I was also on that repo before, switched to this new one because of that error.  The new repo also should have a new gpg key I think.

fcastilloec, 2021-05-03:
@gmipf yes, I checked the installing script so I can manually download the key and manually edit my sources file. I generally don't trust running installation scripts blindly because I don't know what else they're doing to my system. Nonetheless, using the new repo and GPG did fix my problem.

For anybody wondering, you can download the GPG key from here (URL taken from the installation script) https://dl.cloudsmith.io/public/balena/etcher/gpg.70528471AFF9A051.key and try adding it manually.

gmipf, 2021-05-03:
OK tried to add the keyfile manually, here is the output with similar error: 
```
wget -qO - https://dl.cloudsmith.io/public/balena/etcher/gpg.70528471AFF9A051.key | sudo apt-key add -
gpg: invalid key resource URL '/etc/apt/trusted.gpg.d/home:Provessor.gpg'
gpg: keyblock resource '(null)': General error
gpg: key 76F1A20FF987672F: 1 signature not checked due to a missing key
gpg: key 876292BF7F9C32BF: 1 signature not checked due to a missing key
gpg: key 3B4FE6ACC0B21F32: 3 signatures not checked due to missing keys
gpg: key D94AA3F0EFE21092: 3 signatures not checked due to missing keys
gpg: key 871920D1991BC93C: 1 signature not checked due to a missing key
gpg: Total number processed: 17
gpg:       skipped new keys: 17

```

lurch, 2021-05-03:
@gmipf The command you posted works fine for me (on Ubuntu 18.04), maybe there's some other problem with your apt-key configuration? :shrug:

lurch, 2021-05-03:
> using the official Debian repository.

AFAIK the https://deb.etcher.io address was only a redirect to BinTray, which is no longer being used, see #3501 . So I _think_ that means that https://deb.etcher.io is no longer the "official Debian repository"? Perhaps that redirect and/or subdomain ought to be removed @thundron ?

gmipf, 2021-05-03:
OK, I have fixed it now. For whatever reason the already added key "!home:Provessor.gpg" conflicted with the new key installation. It was the key for the following repo which I doesn't need anymore: 
`http://download.opensuse.org/repositories/home:/Provessor/xUbuntu_20.04/`
After removing the key and its repo everything seems to be fine now.

lurch, 2021-05-03:
Perhaps `apt-key` was getting upset/confused by the `!` or the `:` in the filename? :shrug:

closed 2021-05-04 by gmipf