balena-io/etcher #3508: Balena Etcher PPA not recognised

josjoslyn, 2021-05-05:
From 'New deb|rpm PPA #3501' followed instructions as listed for PPA and installation of Balena Etcher on  : https://github.com/balena-io/etcher/issues/3501

My platform/distro is : 

elementary OS 5.1.7 Hera
Built on Ubuntu 18.04.4 LTS
Linux 5.4.0-42-generic
GTK 3.22.30

When running the command as provided to add new PPA it fails as follows :+1: 
jos@Linux-Elementary:~$ curl -1sLf \
>       'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
>       | sudo -E bash
[sudo] password for jos:              
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=elementary  version=5.1.7  codename=hera  arch=x86_64  
 >>>>:
 NOPE: Checking for apt dependency 'apt-transport-https' ...
 NOPE: Updating apt repository metadata cache ...
Failed to update via apt-get update
Get:1 http://ppa.launchpad.net/elementary-os/stable/ubuntu bionic InRelease [15.4 kB]
Get:2 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Hit:3 http://gb.archive.ubuntu.com/ubuntu bionic InRelease
Get:4 http://gb.archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
Hit:5 https://brave-browser-apt-release.s3.brave.com stable InRelease
Hit:6 http://ppa.launchpad.net/elementary-os/os-patches/ubuntu bionic InRelease
Get:7 http://gb.archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
Get:9 http://packages.elementary.io/appcenter bionic InRelease [5,908 B]
Err:8 https://dl.bintray.com/etcher/debian stable InRelease
  403  Forbidden [IP: 3.124.41.190 443]
Get:10 http://ppa.launchpad.net/elementary-os/stable/ubuntu bionic/main Sources [23.5 kB]
Get:11 http://ppa.launchpad.net/elementary-os/stable/ubuntu bionic/main amd64 Packages [34.7 kB]
Get:12 http://security.ubuntu.com/ubuntu bionic-security/main amd64 DEP-11 Metadata [48.5 kB]
Get:13 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 DEP-11 Metadata [60.3 kB]
Get:14 http://ppa.launchpad.net/elementary-os/stable/ubuntu bionic/main i386 Packages [31.7 kB]
Get:15 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 DEP-11 Metadata [2,464 B]
Get:16 http://gb.archive.ubuntu.com/ubuntu bionic-updates/main amd64 DEP-11 Metadata [294 kB]
Get:17 http://gb.archive.ubuntu.com/ubuntu bionic-updates/main DEP-11 48x48 Icons [82.0 kB]
Get:18 http://gb.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 DEP-11 Metadata [289 kB]
Get:19 http://gb.archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 DEP-11 Metadata [2,468 B]
Get:20 http://gb.archive.ubuntu.com/ubuntu bionic-backports/universe amd64 DEP-11 Metadata [9,292 B]
Get:21 http://packages.elementary.io/appcenter bionic/main amd64 DEP-11 Metadata [299 kB]
Reading package lists...
E: Failed to fetch https://deb.etcher.io/dists/stable/InRelease  403  Forbidden [IP: 3.124.41.190 443]
E: The repository 'https://deb.etcher.io stable InRelease' is not signed.
 ^^^^: ... Failed to update via apt-get update - Context above.
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact Cloudsmith (support@cloudsmith.io) for further assistance.
 !!!!: 

Looks - to me - as though it is still looking to 'bintray' for the PPA and getting error :+1: 

Err:8 https://dl.bintray.com/etcher/debian stable InRelease
  403  Forbidden [IP: 3.124.41.190 443]

lurch, 2021-05-05:
Does #3504 help?

lurch, 2021-05-05:
@thundron Looking in detail at the error messages in #3504 and in this issue, I wonder if it only "worked for me" because I already had the `apt-transport-https` package installed (which meant that the cloudsmith setup script didn't try running a (non-working) `apt-get update` before modifying my `/etc/apt/sources.list.d/balena-etcher.list` file) ?

josjoslyn, 2021-05-05:
@lurch Could be the 'why it worked for you'.  My platform is a new build, with no prior apt changes, therefore I would assume (but will verify) that my 'sources.list' should have no reflection/reference to bintray.  That being the case I would have to further assume that the package installer still contains a spurious link ref.

I have just checked my 'sources.list' .... no ref to Balena-etcher
Found in /sources.list.d file /balena-etcher.list - (from failed previous run) now removed

re-run

Added new PPA with 
curl -1sLf \
      'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
      | sudo -E bash

also 
apt-get update
ran -no errors, BUT did not create balena-etcher.list

running the install script fails with E: Unable to locate package balena-etcher-electron

Thoughts anyone

lurch, 2021-05-05:
Just in case you missed the connection - deb.etcher.io is just a redirect to Bintray.

gtoxlili, 2021-05-05:
I found a temporary solution

`curl -1sLf \
   'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
   | distro=ubuntu version=18.04 codename=bionic sudo -E bash`

I feel that this problem occurs because he forgot the elementary OS system~

josjoslyn, 2021-05-05:
@gtoxlili  ...... workaround worked !!!

'c&p' from your entry didn't work, so manually edited the string with the '| distro=...... etc' and it worked, with one entry that reported 'no such file or directory (chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox').  Apart from that finally Balena 1.5.117 is installed and functioning

Many thanks to everyone's input and attention on this matter for me (and no doubt others).  Clearly some more work for you guys to have this working seemlessly.  Clearly multiple distro's gives us great options but headaches for developers of tools like this when many Linux variants have subtle alterations in how installations are achieved.

See output below:

jos@Linux-Elementary:/etc/apt$ curl -1sLf    'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh'| distro=ubuntu version=18.04 codename=bionic  sudo -E bash
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Checking for apt dependency 'apt-transport-https' ...
   OK: Checking for apt dependency 'ca-certificates' ...
   OK: Checking for apt dependency 'gnupg' ...
   OK: Importing 'balena/etcher' repository GPG key into apt ...
   OK: Checking if upstream install config is OK ...
   OK: Installing 'balena/etcher' repository via apt ...
   OK: Updating apt repository metadata cache ...
   OK: The repository has been installed successfully - You're ready to rock!

jos@Linux-Elementary:/etc/apt$ sudo apt install -y balena-etcher-electron
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  gir1.2-vte-2.91 libido3-0.1-0
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  gconf-service gconf-service-backend gconf2 gconf2-common libgconf-2-4
Suggested packages:
  gconf-defaults-service
The following NEW packages will be installed
  balena-etcher-electron gconf-service gconf-service-backend gconf2 gconf2-common libgconf-2-4
0 to upgrade, 6 to newly install, 0 to remove and 1 not to upgrade.
Need to get 87.7 MB of archives.
After this operation, 239 MB of additional disk space will be used.
Get:1 http://gb.archive.ubuntu.com/ubuntu bionic/universe amd64 gconf2-common all 3.2.6-4ubuntu1 [700 kB]
Get:2 http://gb.archive.ubuntu.com/ubuntu bionic/universe amd64 libgconf-2-4 amd64 3.2.6-4ubuntu1 [84.8 kB]
Get:3 https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu bionic/main amd64 balena-etcher-electron amd64 1.5.117 [86.8 MB]
Get:4 http://gb.archive.ubuntu.com/ubuntu bionic/universe amd64 gconf-service-backend amd64 3.2.6-4ubuntu1 [58.1 kB]
Get:5 http://gb.archive.ubuntu.com/ubuntu bionic/universe amd64 gconf-service amd64 3.2.6-4ubuntu1 [2,036 B]
Get:6 http://gb.archive.ubuntu.com/ubuntu bionic/universe amd64 gconf2 amd64 3.2.6-4ubuntu1 [66.9 kB]
Fetched 87.7 MB in 41s (2,130 kB/s)                                                                             
Selecting previously unselected package gconf2-common.
(Reading database ... 183748 files and directories currently installed.)
Preparing to unpack .../0-gconf2-common_3.2.6-4ubuntu1_all.deb ...
Unpacking gconf2-common (3.2.6-4ubuntu1) ...
Selecting previously unselected package libgconf-2-4:amd64.
Preparing to unpack .../1-libgconf-2-4_3.2.6-4ubuntu1_amd64.deb ...
Unpacking libgconf-2-4:amd64 (3.2.6-4ubuntu1) ...
Selecting previously unselected package gconf-service-backend.
Preparing to unpack .../2-gconf-service-backend_3.2.6-4ubuntu1_amd64.deb ...
Unpacking gconf-service-backend (3.2.6-4ubuntu1) ...
Selecting previously unselected package gconf-service.
Preparing to unpack .../3-gconf-service_3.2.6-4ubuntu1_amd64.deb ...
Unpacking gconf-service (3.2.6-4ubuntu1) ...
Selecting previously unselected package gconf2.
Preparing to unpack .../4-gconf2_3.2.6-4ubuntu1_amd64.deb ...
Unpacking gconf2 (3.2.6-4ubuntu1) ...
Selecting previously unselected package balena-etcher-electron.
Preparing to unpack .../5-balena-etcher-electron_1.5.117_amd64.deb ...
Unpacking balena-etcher-electron (1.5.117) ...
Setting up gconf2-common (3.2.6-4ubuntu1) ...

Creating config file /etc/gconf/2/path with new version
Setting up libgconf-2-4:amd64 (3.2.6-4ubuntu1) ...
Setting up gconf-service-backend (3.2.6-4ubuntu1) ...
Setting up gconf-service (3.2.6-4ubuntu1) ...
Setting up gconf2 (3.2.6-4ubuntu1) ...
Setting up balena-etcher-electron (1.5.117) ...
chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox': No such file or directory
Processing triggers for bamfdaemon (0.5.3+18.04.20180207.2-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for libc-bin (2.27-3ubuntu1.4) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Processing triggers for gnome-menus (3.13.3-11ubuntu1.1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for mime-support (3.60ubuntu1) ...
Processing triggers for desktop-file-utils (0.23-1ubuntu3.18.04.2+elementary2~ubuntu5.0.1) ...
jos@Linux-Elementary:/etc/apt$ 

Again many thanks .............

lurch, 2021-05-05:
> with one entry that reported 'no such file or directory (chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox').

Yup, see #3486

revjasper, 2021-05-08:
Looks like Bintray is no more, going to the Etcher page over there shows the following message at the top of the page:
Thanks for supporting Bintray! This service has now been sunset, and to assist with migration to the JFrog Platform, paid accounts can login until July 4th.

I'm going to use the Appimage for now, direct from https://www.balena.io/etcher/ until the repository finds a new home.

lurch, 2021-05-08:
> until the repository finds a new home.

#3501 ? :thinking:

closed 2021-07-07 by thundron