balena-io/etcher #3509: Unable to run Etcher with sandbox or without changing ownership/mode

engineer22, 2021-05-07:
- **Etcher version: 1.5.117** 
- **Operating system and architecture:Debian 10 Buster (amd64)** 
- **Image flashed: None** 
- **Do you see any meaningful error information in the DevTools?: None** 

I followed the documentation for Debian and after executing _sudo apt-get install balena-etcher-electron_ I get the following (look highlighted line):

> user@testsystem:~$ sudo apt-get install balena-etcher-electron
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:                                
  gconf2
The following NEW packages will be installed:                                       
  balena-etcher-electron gconf2
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.                      
Need to get 87.2 MB of archives.
After this operation, 231 MB of additional disk space will be used.                 
Do you want to continue? [Y/n] y
Get:1 http://deb.debian.org/debian buster/main amd64 gconf2 amd64 3.2.6-5 [426 kB]  
Get:2 https://dl.cloudsmith.io/public/balena/etcher/deb/debian buster/main amd64 balena-etcher-electron amd64 1.5.117 [86.8 MB]                                          
Fetched 87.2 MB in 1min 35s (916 kB/s)                                              
Selecting previously unselected package gconf2.                                     
(Reading database ... 214138 files and directories currently installed.)            
Preparing to unpack .../gconf2_3.2.6-5_amd64.deb ...                                
Unpacking gconf2 (3.2.6-5) ...
Selecting previously unselected package balena-etcher-electron.                     
Preparing to unpack .../balena-etcher-electron_1.5.117_amd64.deb ...                
Unpacking balena-etcher-electron (1.5.117) ...                                      
Setting up gconf2 (3.2.6-5) ...
Setting up balena-etcher-electron (1.5.117) ...                                     
**_chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox': No such file or directory_**
Processing triggers for desktop-file-utils (0.23-4) ...                             
Processing triggers for mime-support (3.62) ...                                     
Processing triggers for hicolor-icon-theme (0.17-2) ...                             
Processing triggers for gnome-menus (3.31.4-3) ...                                  
Processing triggers for man-db (2.8.5-2) ...                                        

Trying to run Etcher from the menu, nothing happens. If I go to _/opt/balenaEtcher/_ and do _./balena-etcher-electron_, I get this error:

> [3197:0507/025240.201226:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /opt/balenaEtcher/chrome-sandbox is owned by root and has mode 4755.                                                                    
./balena-etcher-electron: line 10:  3197 Trace/breakpoint trap   "${script_dir}"/balena-etcher-electron.bin "$@"

The SETUID is not configured correctly. If I run _ls -l /opt/balenaEtcher/chrome-sandbox_ I see **-rwxr-xr-x 1 root root  chrome-sandbox** which I think should have the 's' in it like this: **-rwsr-xr-x 1 root root chrome-sandbox**.

### My workaround
I'm not sure if this is the correct way to do it manually and if it's safe (please let me know if this is safe and the supposed action anyway), but by doing the following it works.

I first chown'd the file by `sudo chown root chrome-sandbox` and then chmod'ed it with `sudo chmod 4755 chrome-sandbox` as asked in the error above. Now Etcher works fine, but I'm not sure if the workaround is the supposed action of the setup script and if its safe for my situation.

lurch, 2021-05-07:
Duplicate of #3486 ?

gianlucarenzi, 2021-05-07:
I fixed it creating a symlink in /opt/balena-etcher-electron pointing to /opt/balenaEtcher created by installer.

engineer22, 2021-05-07:
@gianlucarenzi your fix works like a charm.

Did exactly what you said and the installation finished flawlessly. Thank you!

closed 2021-05-09 by engineer22