balena-io/etcher #3510: Etcher gives me a zip file

Funmoring46, 2021-05-08:
- **Etcher version:** v1.5.116
- **Operating system and architecture:** Ubuntu 20.04.2 LTS (64-bit)
- **Image flashed:** None
- **Do you see any meaningful error information in the DevTools?** no

When I download Etcher it gives me a zip file why does it give me that?

Funmoring46, 2021-05-08:
[image]

lurch, 2021-05-08:
The zip file contains an AppImage file, which is what you actually run (after e.g. copying it to your home folder or desktop). IIRC the AppImage gets put inside a `.zip` file because that way the executable-bit of the file is preserved, and it's easier to tell (novice) users to "download the zip file and extract the AppImage" than it is to tell them "download the AppImage and then make it executable".
:shrug: