balena-io/etcher #3511: DEB repo: package list refresh fails, error 402 (payment required)

Miyamoto72, 2021-05-11:
- **Etcher version:** 
latest
- **Operating system and architecture:** 
Debian and derivatives
- **Image flashed:** 
not relevant
- **Do you see any meaningful error information in the DevTools?** 
When refreshing the package lists by issueing 'apt update' I get the following error messsages:

> E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/linuxmint/dists/ulyssa/InRelease  402  Payment Required [IP: 2600:9000:20e8:5a00:e:f4d2:20c0:93a1 443]
> E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/linuxmint ulyssa InRelease' is no longer signed.
> E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/linuxmint/dists/ulyssa/InRelease  402  Payment Required [IP: 2600:9000:20e8:5a00:e:f4d2:20c0:93a1 443]
> E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/linuxmint ulyssa InRelease' is no longer signed.

This makes the repo unusable - maybe cloudsmith has a problem or demands money?

eugeneromero, 2021-05-11:
Can confirm, same issue on Ubuntu 20.04. Additionally, trying to download the repo install script fails with the same (402) error.

lurch, 2021-05-11:
Wow, you've chewed through 200GB of bandwidth already?! (I assume Etcher is on the Open-Source tier? https://cloudsmith.com/product/pricing/ )
I guess Etcher is just **too** popular! :rofl:

lurch, 2021-05-11:
Cloudsmith repo seems to be working again :+1: 
Thanks @thundron

thundron, 2021-05-17:
We're unblocked for now but we're still working towards a "final" solution as the current one is still fairly limited compared to our ideal usage, I'll leave this open for the time being

Frittierapparat, 2021-12-31:
It's happening again. A few days ago I installed it just fine, now I get the same "402 Payment Required".

oguzkarayemis, 2022-01-01:
Same here, apt update returns this error code.

jny986, 2022-01-03:
Having the issue this morning after working fine for a while. 
Pop Os distro.

`E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/pop/dists/hirsute/InRelease  402  Payment Required [IP: 13.226.109.16 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/pop hirsute InRelease' is no longer signed.`

Zorono, 2022-01-03:
The same here too.... Ubuntu 21.10 (AMD64)
```
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/impish/InRelease  402  Payment Required [IP: 54.230.114.10 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu impish InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

mcraa, 2022-01-05:
see #3672

closed 2022-01-05 by mcraa

NunoSempere, 2022-01-08:
Not working for me.

lurch, 2022-01-08:
Yeah, seems to have broken again:
```
Reading package lists... Done                                    
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/bionic/InRelease  402  Payment Required [IP: 99.86.119.33 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu bionic InRelease' is no longer signed.
```

ratajs, 2022-01-09:
I‘m currently encountering this problem on KDE Neon (Ubuntu 20.04).

Josee9988, 2022-10-31:
Same here Ubuntu 22.10

mcraa, 2022-10-31:
It is strange, there is plenty available from the bandwidth for this month.
We'll ping the provider if there was any issue on their side.
Worst case it should reset as the new month starts.
Thanks for the report.

ssi-anik, 2022-11-01:
Seems like the issue is with the provider. I don't use etcher. *symfony cli* uses cloudsmith and i faced the same issue. 

```
E: Failed to fetch https://dl.cloudsmith.io/public/symfony/stable/deb/ubuntu/dists/focal/InRelease  402  Payment Required [IP: 13.227.138.75 443]
```

fermuch, 2023-03-19:
Hello. I'm sorry to revive an old thread, but it is happening again. Should I open a new issue?

shreefgit, 2023-03-20:
Err:11 https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu kinetic InRelease
  402  Payment Required [IP: 18.67.111.84 443]

asterix13, 2023-03-20:
I removed any file with reference to cloud cloudsmith  and balena etcher in the folder "/etc/apt/sources.list.d"

worked for me

LarnuUK, 2023-03-20:
> I removed any file with reference to cloud cloudsmith and balena etcher in the folder "/etc/apt/sources.list.d"
> 
> worked for me

That's not really a "solution", as now once the repository is fixed you won't obtain any updates as you've removed the repo from `apt`. You would have been "better" off disabling the repo for now in my opinion, which can by done by prefixing the repo address in the file with a hash (`#`) or using your package manager. Then once it's confirmed in this thread, or a different one, you can re-enable the repo. You'll instead need to re-add the repository(s) now once the issue is resolved.

asterix13, 2023-03-20:
Oh right, should've thought about that. However I did not need to balena etcher anyways so I removed it out of sheer frustration.

drbornot, 2023-03-20:
I can confirm the issue since yesterday I couldn't fetch the repo for updates

```
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/pop/dists/jammy/InRelease  402  Payment Required [IP: 143.204.89.12 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/pop jammy InRelease' is no longer signed.
```

Argentino84, 2023-03-20:
Same issue on Ubuntu focal.

E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/focal/InRelease 402  Payment Required [IP: 2600:9000:2190:be00:e:f4d2:20c0:93a1 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu focal InRelease' is no longer signed.

lenhatquang2512, 2023-03-20:
same issue on Ubuntu 20.04 LTS.

```
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/focal/InRelease  402  Payment Required [IP: 18.160.249.30 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu focal InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

tsantor, 2023-04-25:
Same issue in Ubuntu 22.04 LTS.

```
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/jammy/InRelease  402  Payment Required [IP: 18.165.32.112 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu jammy InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: https://packagecloud.io/slacktechnologies/slack/debian/dists/jessie/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
```

xarpy, 2023-04-25:
Same issue on Ubuntu 22.04 LTS:

```
E: Falhou ao buscar https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/jammy/InRelease  402  Payment Required [IP: 108.139.134.25 443]
E: O repositório 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu jammy InRelease' já não está assinado.
N: Atualizações a partir de tal repositório não podem ser feitas de forma segura e estão, portanto, desativadas por definição.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: https://packagecloud.io/slacktechnologies/slack/debian/dists/jessie/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.

```

GeanVas, 2023-04-25:
It's on Debian Bullseye too
```
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/debian/dists/bullseye/InRelease  402  Payment Required [IP: 65.8.248.50 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/debian bullseye InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.

```

lurch, 2023-04-25:
See the most recent comments in #4034

aethernet, 2023-04-26:
Hello,

We're moving out of Cloudsmith as continuing to host the package there would be too costly for us.

We're very thankful for the service and help CloudSmith provided, and would recommend their product for smaller open source app and private packages anytime!

At the moment we don't have the bandwidth (pun intended) to set up an alternative repository but might be moving to self hosted (probably CloudFlare R2 + workers) in the upcoming weeks.

So sorry for the inconvenience and thanks you for your patience.

Note that you can always get a .deb and .rpm package in the release section of the github repository.

pmo-19, 2023-04-29:
Unfortunately no I can't since in My Ubuntu Focal 20.04 I have package balena-etcher-electron installed which is not available in your GitHub release section ...

RaibeartRuadh, 2023-05-08:
I confirm that after removal from source list balena-etcher, the update proceeded successfully. The "402 Payment Required" message no appeared.
Ubuntu 22 -> 23

omar55221, 2023-05-16:
a solution for Ubuntu 18.04 was to remove the balena etcher by following step no. 5 
https://www.cyberithub.com/how-to-install-and-use-balenaetcher-on-debian-11-bullseye/#Step_5_Uninstall_balenaEtcher
and added sudo before rm commands

gamesover, 2023-07-17:
@omar55221 it works for me. Thank you