balena-io/etcher #3515: how to find the target

bazghandizahra, 2021-05-12:
- **Etcher version:** v1.5.120
- **Operating system and architecture:**  windows10
- **Image flashed:** ubuntu-20.04.2.0-desktop-amd64.iso 
Hi everyone. I am trying to flash the above iso image in Etcher but I don't know how to find the USB drive to enter as my target. I attached the following image to show what I see when clicking on the target drive here. could you help me with this plz?
[image]

lurch, 2021-05-12:
Looks like Etcher can see your internal hard-drive, but can't "see" your USB drive. Does your USB drive show up in Windows Explorer and/or Device Manager?

bazghandizahra, 2021-05-12:
Thank you for response. The problem was my USB device.thanks for your time

closed 2021-05-17 by thundron