balena-io/etcher #3517: No popup showing that selected ISO file may be a Windows image and should be flashed with another software

xrew11, 2021-05-15:
- **Etcher version:** 1.5.120
- **Operating system and architecture:** Arch Linux x86_64
- **Image flashed:** Windows 10 20H2
- **Do you see any meaningful error information in the DevTools?** No.

https://user-images.githubusercontent.com/73398428/118361463-d7c49b80-b58b-11eb-9310-4f945327bc00.mp4

After selecting a Windows ISO, there's a popup showing that there's a missing partition table instead of the one telling to change the software to flash a Windows image (for example WoeUSB as mentioned in the correct one). Although, the popup is shown when selecting the Windows ISO via url, it's being immediately replaced by itself with missing partition table message while being impossible to read as the whole message is too big to read in that short time (had to take a screenshot to read it fully).

lurch, 2021-05-15:
>  the popup is shown when selecting the Windows ISO via url, it's being immediately replaced by itself with missing partition table message

If you pause your video at _just_ the right place, you can see that this **does** also happen when selecting an ISO image :frowning_face: 
[image]

xrew11, 2021-05-15:
Oh, right, that was so fast that I didn't noticed

thundron, 2021-05-17:
Good catch!