balena-io/etcher #3519: When flash fails there is a drive letter visible but not accessible in this pc(windows).

ADI13579, 2021-05-16:
balenaEtcher-Portable-1.5.120
Windows 10 x64
Zorin-OS-15.3-Core-64-bit.iso

When flash fails there is a drive letter(say F: which was same as the USB drive before flashing the image) visible but windows says it is not accessible in "This Pc", however restarting would resolve the problem.

lurch, 2021-05-16:
Sounds like a "problem" with Windows, rather than a "problem" with Etcher? :thinking: 

> The drive works fine after deleting and creating the partition.

Cool, see also https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

maxmod, 2021-05-19:
diskpart is long process use this small utility to fix pendrive/flash drive https://formatusb.com work well in my case.

closed 2021-06-02 by thundron