balena-io/etcher #3520: Apple M1 - etcher goes blank after picking image

tomaszrondio, 2021-05-21:
- **Etcher version:** 
1.5.120 
- **Operating system and architecture:** 
Big Sur on M1
- **Image flashed:** 
none :)
- **Do you see any meaningful error information in the DevTools?** 
nope. Dev tools get disconnected

rawkitscience, 2021-05-25:
Same issue here, attempted to run without the SD reader installed, no dice.

DarkWarden85, 2021-05-26:
Same issue here, running on a raspberry pi 400 with Raspberry Pi OS.

lurch, 2021-05-26:
For Raspberry Pi, see #2870 
(or you could use [Raspberry Pi Imager](https://www.raspberrypi.org/software/))

2pl, 2021-05-29:
Similar issue here, using version 1.5.120 on macbook M1 / BigSur 11.4
-opening clonezilla-live-2.7.1-22-amd64.zip => blank screen
-opening clonezilla-live-2.7.1-22-amd64.iso => works fine

mattray, 2021-06-01:
Here's the CLI output for that error:
```
$ /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
(node:30189) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app-arm64/generated/modules/xxhash/build/Release/hash.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
(Use `balenaEtcher Helper (Renderer) --trace-warnings ...` to show where the warning was created)
(node:30189) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app-arm64/generated/modules/@balena.io/usb/build/Release/usb_bindings.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
ready-to-show: 306.448ms
Checking for update
Update for version 1.5.120 is not available (latest version: 1.5.120, downgrade is disallowed).
Checking for update
Update for version 1.5.120 is not available (latest version: 1.5.120, downgrade is disallowed).
dyld: lazy symbol binding failed: Symbol not found: _lzma_get_progress
  Referenced from: /Applications/balenaEtcher.app/Contents/Resources/app-arm64/generated/modules/lzma-native/binding/lzma_native.node
  Expected in: flat namespace

dyld: Symbol not found: _lzma_get_progress
  Referenced from: /Applications/balenaEtcher.app/Contents/Resources/app-arm64/generated/modules/lzma-native/binding/lzma_native.node
  Expected in: flat namespace
```

alexclaydon, 2021-07-24:
I'm experiencing this on an M1 Mac.  Reverting to v1.5.114 (an Intel build) resolved the issue.

WiDesignDeveloper, 2021-08-18:
Select the application and "get info", check "open using Rosetta", it will take a while to open the first time but then will work perfectly.

mattray, 2021-08-22:
Can confirm that @WiDesignDeveloper's suggestion works.

DanjelTahko, 2021-09-01:
> Select the application and "get info", check "open using Rosetta", it will take a while to open the first time but then will work perfectly.

Thank you, that works!

HamishMeares, 2021-09-04:
Updating Etcher to Version 1.5.122 worked great on my M1 iMac Big Sur(11.5.2), earlier versions didn't. 
No need for Rosetta, just required some permissions to write to external drive in pop-up window.

thundron, 2021-09-06:
@HamishMeares that is extremely interesting given we got a report here https://github.com/balena-io/etcher/issues/3593 saying that 1.5.122 lost the "universal" and apparently the arm64 file definitions? :/

thundron, 2021-09-06:
we're investigating in that report anyway, so hopefully we'll have more to share either here or there

closed 2023-07-12 by dfunckt