balena-io/etcher #3521: Etcher not working on Debian 64bit

valdominic, 2021-05-22:
I have been trying to run etcher from the last 4 months but i am stuck with the Debian OS because i can't get Etcher to run. I have seen another post of a user having the issue and i tried following the install instructions that was already available but Etcher does not work. 

Links that I used:-
https://forums.balena.io/t/balenaetcher-1-58-89-app-image-cant-be-invoked/115243/25
https://github.com/balena-io/etcher
https://www.electronjs.org/apps/etcher

I think there is a serious problem on Debian for BalenaEtcher. I never had any issues with other distros that i have used like Arch, Ubuntu, OpenSuse. Yes, I keep trying out new distros.

Please help.

lurch, 2021-05-24:
Duplicate of #3486 ?

valdominic, 2021-05-24:
> Duplicate of #3486 ?

Thank you @lurch. Yes it is a duplicate. I don't think this is fixed yet.

closed 2021-05-31 by thundron

BTCGandalf, 2022-05-12:
I am having this issue as well trying to install Balena Etcher on Kubuntu.

I can get it installed via terminal but it won't launch (the icon appears in the task bar but then dissappears and Etcher never runs).

If I try to download and run the app image nothing happens (executable box is ticked)

valdominic, 2022-05-13:
I agree it doesn't work on Kubuntu. I had the exact same issue, it gets
installed without any issues but doesn't run when you try to run it.

lurch, 2022-06-12:
Apparently a fix is being worked on... https://github.com/balena-io/etcher/issues/3771#issuecomment-1152065852

bfe0503, 2022-12-06:
does not work all I have tried does not help. Etcher was a tool which make it easy, but now it is one more not usable tool like millions out there. 
Sorry, user don't need workarounds, they need solutions.