balena-io/etcher #3523: Failure to handle http 'Location: ' redirects results in "Missing Partition Table" error messages

deftdawg, 2021-05-30:
- **Etcher version:** 1.5.120
- **Operating system and architecture:**  Windows and OSX
- **Image flashed:** Flash from URL: https://files.kde.org/neon/images/unstable/current/neon-unstable-current.iso
- **Do you see any meaningful error information in the DevTools?**  No

Requests to `https://files.kde.org/neon/images/unstable/current/neon-unstable-current.iso` are redirected to different mirrors; in this request it's directed to `https://mirror.its.dal.ca/kde-applicationdata/neon/images/unstable/20210530-0328/neon-unstable-20210530-0328.iso`

In all cases etcher displays `Missing Partition Table` and if continue is selected flashes a '0' size image.  If the image is downloaded from a location in a browser, etcher is happy to flash it without complaints and does it correctly.

Expected behaviour would be to follow redirects and then flash that.

```
> GET /neon/images/unstable/current/neon-unstable-current.iso HTTP/2
> Host: files.kde.org
> user-agent: curl/7.75.0
> accept: */*
>
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [57 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [57 bytes data]
* old SSL session ID is stale, removing
{ [5 bytes data]
* Connection state changed (MAX_CONCURRENT_STREAMS == 100)!
} [5 bytes data]
< HTTP/2 302
< date: Sun, 30 May 2021 22:03:54 GMT
< server: Mirrorbits/v0.5.1-30-gc5f884e-dirty
< cache-control: private, no-cache, max-age=0
< content-type: text/html; charset=utf-8
< link: <https://mirror.its.dal.ca/kde-applicationdata/neon/images/unstable/20210530-0328/neon-unstable-20210530-0328.iso>; rel=duplicate; pri=1; geo=ca
< location: https://mirror.math.princeton.edu/pub/kde/files/neon/images/unstable/20210530-0328/neon-unstable-20210530-0328.iso
< content-length: 0
```

Nmatt1, 2022-01-23:
From what this looks like and what you have said, it appears that Etcher cannot read the partition table either fast enough from the file location or it cannot read it at all. So, when you fed Etcher the file locally it didn't have any time limits and it worked without any issues.

TL;DR This appears to just be a server issue. Ether can't read fast enough, download ISO do it that way.

deftdawg, 2022-01-23:
> TL;DR This appears to just be a server issue. Ether can't read fast enough, download ISO do it that way.

In what way is this a server issue?  

The HTTP spec allows for redirection but does not specify a requirement for a minimum download speed; it's on the client to follow the spec and deal with the speed of transfer if it's expecting to do something with the data.  Etcher (or whatever http client library is using) is responsible for dealing with that if it intends to offer http as an option.

VesaVainio, 2025-06-02:
I have encountered the same issue. I'm integrating Balena Etcher with an application that should provide the image to be flashed. The application endpoint is returning a Redirect response with a Location header pointing to the actual image. In my case the redirect location is located in Azure Blob Storage, which should be about as efficient and scalable as you can get.

It's clear that it is not a server/backend issue. The Etcher just doesn't follow the redirect. It fails immediately, it doesn't even try to start downloading.

I'm guessing that whatever platform you use would have a feature for automatically following redirects. Most HTTP client implementations do. You probably just need to enable it by setting "followRedirects" or something like that to "true" somewhere. Could be a one-liner fix.

Please get this fixed?