balena-io/etcher #3526: Checksum doe not match

ZurMaD, 2021-06-03:
- **Etcher version:** 1.15.117
- **Operating system and architecture:**  Ubuntu 21.04 amd64
- **Image flashed:** Ubuntu 21.04 ISO
- **Do you see any meaningful error information in the DevTools?** 

[image]

thundron, 2021-07-05:
it means some data on the target is different from the source; could be the target itself has corrupted sectors or the reader/slot you used is faulty

closed 2021-07-05 by thundron

pablodz, 2021-07-05:
I suspect that it's the type of USB connection, I failed trying many times with multiple SO versions and USB slots (3.1+) (I don't have USB3.0 or 2.0)

Well, something happens ... 
If any other person read this, you should buy a new USB that supports USB3.2+ native, as I did.

lurch, 2021-07-05:
Unless there's buggy firmware or drivers in the mix somewhere, I _suspect_ that the **type** of USB connection doesn't matter, and it's actually the device hanging off the end of the USB connection (e.g. SD card, HDD, SSD or flash drive) that makes all the difference.
After all the "U" in USB stands for "Universal", and it's explicitly designed to be both forwards and backwards compatible (i.e. you can use a USB2.0 device with a USB3.2 slot, or a USB3.2 device with a USB2.0 slot - each end of the connection negotiates the maximum speed it supports).