balena-io/etcher #3528: Wrong permissions on file chrome-sandbox in Debian repo

PhilipLoewen, 2021-06-06:
- Etcher 1.5.117
- Debian 10 buster x86_64

I added the recommended DEB repository and used it to grab the lastest package. Here are the operative lines from my file named /etc/apt/sources.list.d/balena-etcher.list:
```
deb https://dl.cloudsmith.io/public/balena/etcher/deb/debian buster main
deb-src https://dl.cloudsmith.io/public/balena/etcher/deb/debian buster main
```
Etcher declined to start, saying, 
> [4586:0606/084108.012944:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /opt/balenaEtcher/chrome-sandbox is owned by root and has mode 4755.                                                                                               
./balena-etcher-electron: line 10:  4586 Trace/breakpoint trap   "${script_dir}"/balena-etcher-electron.bin "$@"                                                                             

I managed to solve this problem by giving the following command:
```
 $ sudo chmod 4755 /opt/balenaEtcher/chrome-sandbox 
```
Now Etcher works for me and I am on my way, happy and grateful. Just wanted to let the team know that there may be a packaging issue here.

(Related, perhaps secondary: The quoted message appears when Etcher is launched from the command line. I did that because trying to launch Etcher from the graphical menu system just appeared to do nothing at all. Presumably it was aborting for the same reason ... but a GUI user would never know.)

Thanks for building and sharing this!

lurch, 2021-06-06:
Duplicate of #3129 ?

PhilipLoewen, 2021-06-07:
Yes, looks like the same issue raised in #3129. Thanks for pointing this out. In defense of my submission, I note

1. a lower emotional temperature, and
2. explicit descriptions of a workaround that might point to a potential solution.

My rosy interpretation of "duplicate" status: just think of how many users would rejoice when someone fixes this! :)

closed 2021-06-10 by thundron