balena-io/etcher #3531: error displayed in UI

waLplanet, 2021-06-08:
- **Etcher version:**  Versión 1.5.109 (1.5.109)
- **Operating system and architecture:** macOS 11.4, intel
- **Image flashed:** deepin-desktop-community-20.2.1-amd64.iso

an error displayed in UI (screenshot attached) while trying to update. error response seems to be not handled properly
[image]

thundron, 2021-07-05:
have you been able to reproduce this issue? seems odd to see that error, as the response should only be shown with a response code 200

waLplanet, 2021-07-06:
the response should be status 200, but in real life not 100% time works like that, actually, the server was down when it happened to me.
so, you need to handle possible errors.

thundron, 2021-07-06:
That's not what I said though - I said that the response should be shown only when the code is 200, which means error should not be shown, which is why I was interested in how reproducible this issue was. Looks like it isn't though, closing for the time being

closed 2021-07-06 by thundron

lurch, 2021-07-06:
@thundron Is it possible that the server was working (and returning response code 200) when the flashing process started, and then only returned response code 503 part-way _during_ the flashing process? :thinking:

thundron, 2021-07-07:
@lurch that's why I asked, but we'll see in the future I guess 😉