balena-io/etcher #3532: Support WIN ISOs [RUFUS doesn't work]

jredfox, 2021-06-09:
Windows 7 and xp isos I would like support for

Windows 7 iso
https://download.microsoft.com/download/0/6/3/06365375-C346-4D65-87C7-EE41F55F736B/7601.24214.180801-1700.win7sp1_ldr_escrow_CLIENT_PROFESSIONAL_x64FRE_en-us.iso

Windows xp iso
https://archive.org/details/WinXPProSP3x86

I love etcher's and tried it on windows 7 iso then it says it's not bootable it try's to boot then doesn't go past the windows loading screen just like other programs. I tested it on a usb 1.0 on a usb 2.0 port from a windows 7 computer. and it shouldn't matter because even if the drivers for 3.0 are not installed it will still treat the 3.0 like either a 1.0 or 2.0

**Why not the other programs you linked:**
- I tried the link to rufus's program like your github mentioned and that doesn't really work. I tested it on a pc that previously had windows 7 and wouldn't go past the loading screen. I tried every configuration possible spent 12 hours using both partition types on the windows 7 and modern hardware with no luck.
- I tried anyburn doesn't even show up as a boot option
- windows 7 usb tool creator probably only works for windows 7 on windows 7 and is no longer downloadable

Why?: I am a java software dev trying to setup dev for older os's that people still use for debugging purposes if they have windows 7/xp still and the code should be universal across them when it's not.

lurch, 2021-06-09:
Duplicate of #210 ?

jredfox, 2021-06-09:
yes but, please work on this and add my info to there as it's helpful. I also found x64 iso for xp idk if it's modified or not though. then you can close this issue

lurch, 2021-06-09:
I don't work on Etcher myself, I'm just a community member that tries to be helpful :smiley:

jredfox, 2021-06-09:
well leaving it open as this info will help them solve it alot faster. they should be like ok... here are the isos, I have a usb, here is the ide now what's wrong with it

support-monkey, 2021-06-21:
Hello. i don't even know, why you even bother to refer to Windows XP anymore. It is long since Retired and deprecated. It is long since abandoned. It belongs to history museums now. No one ever cares about Windows XP or even consider it. My friendly advice is move forward and set the minimum OS version you develop for as Windows 8, because this is the earliest supported OS, and ignore the any OS older than that.

jredfox, 2021-06-21:
Actually a lot of school computers and pcs built in 200-2006 still run xp and don’t want to run windows 7 because it would be too much more ram usage. Windows 10 wouldn’t work either

support-monkey, 2021-06-22:
Windows 11 i behind the corner, and we are still speaking about Windows Xp? Such die hard xp people are progressively left behind, no more fixes, no more patches no more features, following outdated protocols, comply to outdated standards, OS unable to actually use modern'hardware's enhanced security features to improve their own security. And memory consumption? I prefer to have few frequently used libraries in memory all the time, so it is accessed instantly instead of constant waste of time to load/unload to save a meg or two. Not to mention SSD trimming and optimizing support for new os vs none in old os. Xp belongs to history museum, or in hobby rigs of retro communities.

jredfox, 2021-06-23:
- java devs code that supports java 8 should support xp as well.
- windows 11 is fake I looked at the videos and saw several things that lead me to believe it was just a modified windows 10.
- my issue focus is on windows 7 not xp. but, xp should also be supported by a utility which brags about being able to work with almost any os which includes but not limited to windows 7 and xp.

lurch, 2021-06-23:
> my issue focus is on windows 7 not xp. but, xp should also be supported by a utility which brags about being able to work with almost any os which includes but not limited to windows 7 and xp.

The [README](https://github.com/balena-io/etcher#supported-operating-systems) says it runs on "Microsoft Windows 7 and later" - but note that this is the OS it _runs_ on which doesn't necessarily mean it can "flash" that OS too.

support-monkey, 2021-06-23:
@jredfox, stay in the past if you so much want. I really don't care about old things. Just make sure you understand i don't want to be involved in pointless discussions about old things went out of support, being retired and deprecated. If you want to waste your time, fine it is yours to waste as you like. Make sure you don't waste my time.

jredfox, 2021-06-23:
Well good thing you are not the dev for etcher otherwise the issue would have been closed. Widnows 7 should work with etcher by default

thundron, 2021-07-05:
please refer to https://github.com/balena-io/etcher/issues/210

closed 2021-07-05 by thundron