balena-io/etcher #3533: ailed install on Knoppix 9.1

jothimurugan2017, 2021-06-11:
- **Etcher version:** 
- Current Version
- **Operating system and architecture:** 
- Linux- Knoppix
- AMD A8 7410 with Radeon R5 

- **Image flashed:** 
- No
- **Do you see any meaningful error information in the DevTools?** 
- root@Microknoppix:/etc/apt/sources.list.d# curl -1sLf \
   'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
   | sudo -E bash
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=debian  version=<empty>  codename=bullseye  arch=x86_64  
 >>>>:
 NOPE: Checking for apt dependency 'apt-transport-https' ...
 NOPE: Updating apt repository metadata cache ...
Failed to update via apt-get update
Hit:1 http://ftp.de.debian.org/debian stable InRelease
Hit:2 http://security.debian.org stable/updates InRelease
Ign:3 http://ftp.debian.org/debian wheezy InRelease
Hit:4 http://ftp.de.debian.org/debian testing InRelease
Get:5 http://ftp.org/debian wheezy InRelease
Hit:6 http://ftp.de.debian.org/debian unstable InRelease
Err:7 http://ftp.debian.org/debian wheezy Release
  404  Not Found [IP: 151.101.154.132 80]
Err:5 http://ftp.org/debian wheezy InRelease
  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
Hit:8 http://ftp.de.debian.org/debian experimental InRelease
Ign:9 http://ftp.de.debian.org/debian oldstable InRelease
Hit:10 http://ftp.de.debian.org/debian stable-updates InRelease
Hit:11 http://ftp.de.debian.org/debian oldstable Release
Reading package lists...
E: The repository 'http://ftp.debian.org/debian wheezy Release' does not have a Release file.
E: Failed to fetch http://ftp.org/debian/dists/wheezy/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: The repository 'http://ftp.org/debian wheezy InRelease' is not signed.
 ^^^^: ... Failed to update via apt-get update - Context above.
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact Cloudsmith (support@cloudsmith.io) for further assistance.
 !!!!:

jothimurugan2017, 2021-06-11:
32bit Appimage works

closed 2021-06-11 by jothimurugan2017

lurch, 2021-06-11:
> E: The repository 'http://ftp.debian.org/debian wheezy Release' does not have a Release file.

https://wiki.debian.org/DebianWheezy says "Wheezy received Long-Term-Support until 2018-05-31. Now it is archived, and no longer receives official security updates. It was superseded by Debian/Jessie on 2015-04-25."

Although https://www.knopper.net/knoppix/knoppix910-en.html says it's based on Debian bullseye, so I wonder why it's looking for wheezy update files? :shrug: