balena-io/etcher #3534: Don't ask me to `curl | bash` as root!

dflock, 2021-06-18:
The README says that Debian/Ubuntu users should do this to install the new deb repo:

```
$ curl -1sLf \
   'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
   | sudo -E bash
```

Please don't ask me to curl things of the internet into `sudo bash` - not a great idea.

So, I downloaded that 14kb script and had a look at it. It's huge and unnecessary and does icky stuff - like messing with system python etc...

Instead of that monster, why not this instead:

```console
$ curl -1sLf "https://dl.cloudsmith.io/public/balena/etcher/config.deb.txt?distro=$(lsb_release -is)&codename=$(lsb_release -cs)&version=$(lsb_release -rs)&arch=$(uname -m)" \
| sudo tee /etc/apt/sources.list.d/balena-etcher.list
```

In both cases you're asking the user to copy & run a single command. The difference is that the second version just directly creates the .list file, without needing the script, uses the least privilege and is auditable by the user.

I've tested this on Ubuntu 20.04 Focal, and it uses standard tools and methods that should work across Debian & Ubuntu releases.

lurch, 2021-06-18:
I'm only a 3rd-party observer and not an Etcher (or Balena) developer, but I wonder if these are questions for cloudsmith.io rather than balena.io ? :man_shrugging: 

Your "alternative one-liner" is much longer, and has _much_ more potential for typos (especially with the multiple `$(...)` parts) if people are typing it by hand rather than directly copy'n'pasting?

dflock, 2021-06-18:
> Your "alternative one-liner" is much longer, and has much more potential for typos (especially with the multiple $(...) parts) if people are typing it by hand rather than directly copy'n'pasting?

True, but I imagine that 99% of people would copy & paste either command - they're both pretty long - why would you type either of them in?

A slightly simpler and more direct option would be something like this - don't request the contents of the .list file from the remote server, just create it directly:

```console
$ echo "deb https://dl.cloudsmith.io/public/balena/etcher/deb/$(lsb_release -is) $(lsb_release -cs) main" \
| sudo tee /etc/apt/sources.list.d/balena-etcher.list
```

lurch, 2021-06-19:
> why would you type either of them in?

I occasionally need to do that if I've got one system running in text-only mode, and I'm running a graphical web-browser on another device. But in hindsight Etcher _requires_ a GUI, so you'd only ever install it on a system **not** running in text-only mode, in which case you can just run a web-browser on the same device that you're going to be installing Etcher to :slightly_smiling_face: 
So yeah, apologies for my irrelevant comment!

dflock, 2021-06-22:
Or maybe just make it separate commands, presented/documented something like this:

-----------------

Add the Balena GPG key:

```console
$ curl -1sLf "https://dl.cloudsmith.io/public/balena/etcher/gpg.70528471AFF9A051.key" | sudo apt-key add -
```

then add the official Balena repository:

```console
$ sudo apt-add-repository "deb https://dl.cloudsmith.io/public/balena/etcher/deb/$(lsb_release -is) $(lsb_release -cs) main"
```

Update & install Balena Etcher:

```console
$ sudo apt update && sudo apt install balena-etcher-electron
```

skipperTux, 2022-01-09:
Cloudsmith supports this, see [Set Me Up](https://cloudsmith.io/~balena/repos/etcher/setup/#formats-deb) for Debian and RedHat.
> **or** ... you can manually configure it yourself before installing packages:

Maybe it should be made more prominent in the Balena Etcher Docs.