balena-io/etcher #3536: White background without any GUI elements, when running Etcher

FroztOfficial, 2021-06-21:
- **Etcher version:** 1.5.120
- **Operating system and architecture:** Arch Linux
- **Do you see any meaningful error information in the DevTools?**
It shows this error in the console, when running the program: "Uncaught Error: /usr/lib/balena-etcher/generated/modules/xxhash/build/Release/hash.node: undefined symbol: _ZM2v86Object18InternalFieldCountEv"

Hey, so when I try to run balena etcher on Arch Linux, it does not load any of the GUI it just shows a white background, which maybe is because that it doesn't load it, because of the error from above.

I don't know if this would be helpful, but the program have been installed from the AUR, which you can find here:
https://aur.archlinux.org/packages/balena-etcher/

oberon-manjaro, 2021-06-22:
Error I am seeing on Manjaro:

`libva error: /usr/lib/dri/i965_drv_video.so init failed
`
version installed is **libva 2.12.0**

oberon-manjaro, 2021-06-22:
The issue seems to be with electron 13.1.2
After downgrading electron to 12.0.9, etcher is working for me again.
Launching from command line I realized that the libva error message is present anyway, but the UI is working again. So likely this was present before ... ;)

sebastian-65, 2021-07-09:
Also affected (Arch).

hockeymikey, 2021-07-16:
How do you get etcher to use the electron12 install? I tried an alias but no luck, still is using 13.

madc, 2021-08-03:
> How do you get etcher to use the electron12 install? I tried an alias but no luck, still is using 13.

On Arch Linux you can use a helper like [downgrade](https://aur.archlinux.org/packages/downgrade/).

senthilmk, 2021-08-03:
Same deal with Cent-os 7.

hockeymikey, 2021-08-03:
> > How do you get etcher to use the electron12 install? I tried an alias but no luck, still is using 13.
> 
> On Arch Linux you can use a helper like [downgrade](https://aur.archlinux.org/packages/downgrade/).

I don't want to downgrade. How do I get etcher to use 12 and only 12 when it's installed but keeping 13.

cyber-loli, 2021-08-07:
> I don't want to downgrade. How do I get etcher to use 12 and only 12 when it's installed but keeping 13.

You can install electron 12 (`pacman -S electron12`) and launch etcher: `electron12 /usr/lib/balena-etcher`

FroztOfficial, 2021-08-13:
> You can install electron 12 (`pacman -S electron12`) and launch etcher: `electron12 /usr/lib/balena-etcher`

Thanks, this seems to work for me.
We will just have to use this work around until they fix the issue for electron 13.

thundron, 2021-08-17:
Duplicate of https://github.com/balena-io/etcher/issues/3536

thundron, 2021-08-17:
There was a bit of confusion, I meant to link to https://github.com/balena-io/etcher/issues/3520 and not close (yet) since they could be different issues (the other one is for Apple M1 chip)

thundron, 2021-08-17:
Keep in mind that the AUR packages are community-driven, so you're better off checking the issue there. Apparently an user got it working under a different AUR here https://github.com/balena-io/etcher/issues/3558#issuecomment-886866263 so if you can check that, we could close this issue for now as we're not using Electron 13 on master yet

galaxine-senpai, 2022-12-29:
I am having the same issue on Zorin, here is my dev tools console output:

`
Uncaught Error: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/.mount_balenanLSKvr/resources/app/generated/modules/@ronomon/direct-io/binding.node)
    at process.func [as dlopen] (electron/js2c/asar_bundle.js:5)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1138)
    at Object.func [as .node] (electron/js2c/asar_bundle.js:5)
    at Module.load (internal/modules/cjs/loader.js:935)
    at Module._load (internal/modules/cjs/loader.js:776)
    at Function.f._load (electron/js2c/asar_bundle.js:5)
    at Function.o._load (electron/js2c/renderer_init.js:33)
    at Module.require (internal/modules/cjs/loader.js:959)
    at require (internal/modules/cjs/helpers.js:88)
    at Object.<anonymous> (gui.js:1)
`

lurch, 2022-12-29:
@galaxine-senpai That _might_ have been fixed by #3935 ?

VincentJGeisler, 2023-06-02:
running into the same problems on windows portable 1.18.4

jankaifer, 2023-10-14:
Checking in here from NixOs https://github.com/NixOS/nixpkgs/pull/158715. NixOS still has etcher 1.7.9 with electron 12 which is quite old now.

Do you still guys support only electron 12 or is there any new version that we could use?

DavidAntliff, 2024-03-13:
White screen here, with Ubuntu 22.04 + Etcher 1.18.11 when run with `sudo` (`sudo` needed due to missing polkit thingie).

1.8.0 does not have this problem, so will use this for now.