balena-io/etcher #3537: Application not opening

MasterDiggi, 2021-06-22:
I installed Etcher normally, but when I try to execute it nothing happens even tho it's an active process

diablos6p, 2021-07-04:
can you post the terminal output of those two commands.
ls -l | grep balena
sudo ls -l /opt | grep balena
I had the same problem and it turned out that there was a problem with the shortcut link.

lurch, 2021-07-04:
@MasterDiggi What OS are you running on? What version of Etcher are you using?

MasterDiggi, 2021-07-13:
> can you post the terminal output of those two commands.
> ls -l | grep balena
> sudo ls -l /opt | grep balena
> I had the same problem and it turned out that there was a problem with the shortcut link.

I'm not using Linux, I'm using windows

MasterDiggi, 2021-07-13:
> @MasterDiggi What OS are you running on? What version of Etcher are you using?

I'm using Windows 10 Pro and I'm using the latest version of Etcher

MasterDiggi, 2021-07-13:
this is how the app stays - https://imgur.com/a/GuFCBvd

lurch, 2021-07-13:
Sounds very similar to #3088 ?

diablos6p, 2021-07-18:
did you try to use an older version of the app?
you can find them here: https://github.com/balena-io/etcher/releases

MasterDiggi, 2021-07-27:
> did you try to use an older version of the app?
> you can find them here: https://github.com/balena-io/etcher/releases

ok yes I've tried using old versions and the only place I could use it was on my Linux laptop and when I restart my pc trying to open the app as fast as possible

thundron, 2021-08-20:
sounds like a duplicate of #3088 as mentioned, closing this one

closed 2021-08-20 by thundron